OpenTabletDriver/OpenTabletDriver #2524: `OpenTabletDriver` cannot be built on iOS platform

frenzibyte, 2022-12-07:
### Acknowledgements

- [ ] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Steps to Reproduce

1. Create a .NET 6 iOS project
2. Add reference to `OpenTabletDriver` package
3. Build project

### Expected Behavior

Project builds successfully.

### Observed Behavior

Project fails building during linking stage:

[image]

---

This is happening as a result of `OpenTabletDriver` referencing `OpenTabletDriver.Native` which includes P/Invokes to the `Quartz` framework. This causes dotnet to attempt including the framework during the linking stage, despite that framework being macOS-only.

### Tablet Device

–

### Diagnostic Information

–

32-bit-link, 2022-12-07:
+1 on this personally. Having OpenTabletDriver support on iOS would be really useful.

frenzibyte, 2022-12-07:
One solution I can think of would be to move the macOS P/Invokes to a separate project that's referenced by `OpenTabletDriver.Desktop` directly instead, and while it's *somewhat* ugly, it completely fixes the issue for me: https://github.com/OpenTabletDriver/OpenTabletDriver/compare/master...frenzibyte:OpenTabletDriver:separate-macos-pinvokes (@X9VoiD thoughts on this would be appreciated).

AkiSakurai, 2022-12-07:
There is a workaround on xamarin issues tracker.

https://github.com/xamarin/xamarin-macios/issues/15118#issuecomment-1141893683

gonX, 2026-10-04:
I'm reopening because this code has never reached a release. osu-framework also has a workaround in place:
https://github.com/ppy/osu-framework/blob/254663758b77f0b5e64ccdd5e8e60d0be9044549/osu.Framework.iOS/osu.Framework.iOS.Workarounds.targets#L20-L29

This seems to have been closed as fixed because it was fixed on `master` and `avalonia` but our releases do not use code from these branches yet.

PR #2528 (2022-12-14): Split OpenTabletDriver.Native to per platform
Also some smaller stuff:

- Add a reusable generic `pack_local.sh`
- Added `.local_feed` to .gitignore, more on this later.
- Explicitly only include `OpenTabletDriver.Native.Windows` in `OpenTabletDriver.csproj`.
- Rename `OpenTabletDriver.Native.OSX` namespace to `OpenTabletDriver.Native.MacOS`.

Fixes #2524

Original idea is thanks to @frenzibyte!

### Usage of `pack_local.sh`

```
Usage: pack_local.sh <options> [pack|remove] [project path/csproj file]
Options: -h|--help
         -f|--feed <feed path> (default: ./.local_feed)
```

The changes to [... cut]