OpenTabletDriver/OpenTabletDriver #2532: Add support for HUION RDS-160

Zmuogs, 2022-12-10:
### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Manufacturer

HUION

### Product Name

RDS-160

### Device Width

344.16 mm

### Device Height

193.6 mm

### Pressure Levels

8192

### Lines per Inch (LPI)

5080

### Additional Information

Connected using "3in1" cable (USB for control, HDMI for display and USB for power).

Also tested under Arch using official "huiontablet" driver from AUR. The pen works, the tablet keys also work, but UI is buggy and incomplete.

### Diagnostic Information

[diagnostics.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/10200816/diagnostics.txt)

jamesbt365, 2022-12-14:
Downgrade to the latest release and remove the settings file in `~/.config/OpenTabletDriver` then:

tablet > device string reader
9580
109
leave it blank
dump all. Save that file and send it here.

Zmuogs, 2022-12-15:
Downgraded to "opentabletdriver 0.6.0.4-3".
File: [device_string_reader__dump_all.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/10235210/device_string_reader__dump_all.txt)

jamesbt365, 2023-04-07:
Could you follow https://opentabletdriver.net/Wiki/FAQ/Linux#argumentoutofrangeexception and export diagnostic again so we can proceed?

Zmuogs, 2023-04-20:
On a another computer (laptop):
- executed commands from FAQ,
- installed OpenTabletDriver v0.6.0.4,
- just in case, restarted the computer,
- created the dump: 
[dump__kubuntu_22.04__OpenTabletDriver_0.6..0.4.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/11285998/dump__kubuntu_22.04__OpenTabletDriver_0.6.0.4.txt)

jamesbt365, 2023-08-23:
Still need a diagnostic to tell which interface to use when creating the configuration. @Zmuogs

Zmuogs, 2023-11-17:
Sorry for delay.

In v0.6.0.4 I  always got "Not permitted to open HID class device" messages no mater what I did.
Upgraded to v0.6.3.0, and now it recognizes my tablet.

There is outputs:
[diagnostics.json](https://github.com/OpenTabletDriver/OpenTabletDriver/files/13390390/diagnostics.json)
[string_reader.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/13390392/string_reader.txt)
[dmesg.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/13390393/dmesg.txt)
[ls_l_dev_hidraw.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/13390395/ls_l_dev_hidraw.txt)

jamesbt365, 2023-12-26:
Fixed by #2786.

closed 2023-12-26 by jamesbt365