OpenTabletDriver/OpenTabletDriver #2538: Add support for Huion Kamvas 22 [Not Plus]

dinispimentel, 2022-12-17:
### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Manufacturer

Huion

### Product Name

Kamvas 22

### Device Width

546mm

### Device Height

323mm

### Pressure Levels

8192

### Lines per Inch (LPI)

5080

### Additional Information

Connected using "3in1" cable (USB for control, HDMI for display and USB for power).

Also tested under Arch using official "huiontablet" driver from AUR. The pen works, the tablet keys also work, but UI is buggy and incomplete.

### Diagnostic Information

[diagnostics.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/10252292/diagnostics.txt)

jamesbt365, 2022-12-18:
tablet > device string reader
9580
109
leave blank
dump all, save that file as a .txt and send it here

dinispimentel, 2022-12-19:
I tried to use the Device String Reader but it keeps asking to re-plug the tablet when I click Dump All.

I re-checked the fields, plugged, re-plugged and same.

That said, I'll send the lsusb verbose output of the device (256c:006d [effectively VID: 9580 | PID: 109]).

I don't know if it is useless. If it is, I'm sorry, and I would appreciate some support to proceed the 'Dump All' action.
[lsusb-verbose-dump.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/10255601/lsusb-verbose-dump.txt)

jamesbt365, 2023-04-07:
Unfortunately this doesn't provide me with the device strings, you may need to use windows to get the strings properly, or try another tool for linux.

jamesbt365, 2025-01-24:
Fixed by #2921.

closed 2025-01-24 by jamesbt365