OpenTabletDriver/OpenTabletDriver #2547: Support Request:

holahapi, 2022-12-22:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### What do you need help with?

can not detect tablet GM156HD

### What have you tried?

install .net runtime 6.0.12
uninstall gaomon, wacom by Settings app or Control Panel
ran WinUSB.cleanup.script\cleanup.bat

### Diagnostic Information

{
  "App Version": "OpenTabletDriver v0.6.0.4",
  "Build Date": "5/4/2022",
  "Operating System": {
    "Platform": 2,
    "ServicePack": "",
    "Version": "10.0.19043.0",
    "VersionString": "Microsoft Windows NT 10.0.19043.0"
  },
  "Environment Variables": {
    "USER": null,
    "TEMP": "C:\\Users\\Michael\\AppData\\Local\\Temp",
    "TMP": "C:\\Users\\Michael\\AppData\\Local\\Temp",
    "TMPDIR": null,
    "USERPROFILE": "C:\\Users\\Michael"
  },
  "HID Devices": [
    {
      "DevicePath": "\\\\?\\hid#vid_046d&pid_c539&mi_01&col04#7&6820818&0&0003#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "Manufacturer": "Logitech",
      "ProductName": "USB Receiver",
      "SerialNumber": "",
      "FriendlyName": "USB Receiver",
      "VendorID": 1133,
      "ProductID": 50489,
      "InputReportLength": 2,
      "OutputReportLength": 0,
      "FeatureReportLength": 0,
      "CanOpen": true
    },
    {
      "DevicePath": "\\\\?\\hid#vid_2717&pid_5010&mi_01&col01#7&277bb8f4&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "Manufacturer": "MOSART Semi.",
      "ProductName": "Mi Wireless Combo",
      "SerialNumber": "",
      "FriendlyName": "Mi Wireless Combo",
      "VendorID": 10007,
      "ProductID": 20496,
      "InputReportLength": 7,
      "OutputReportLength": 0,
      "FeatureReportLength": 0,
      "CanOpen": false
    },
    {
      "DevicePath": "\\\\?\\hid#vid_2717&pid_5010&mi_01&col04#7&277bb8f4&0&0003#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "Manufacturer": "MOSART Semi.",
      "ProductName": "Mi Wireless Combo",
      "SerialNumber": "",
      "FriendlyName": "Mi Wireless Combo",
      "VendorID": 10007,
      "ProductID": 20496,
[... 1750 lines cut ...]
      "Time": "2022-12-22T15:26:34.3763207+08:00",
      "Group": "Display",
      "Message": "DPI Awareness enabled",
      "StackTrace": null,
      "Level": 0,
      "Notification": false
    },
    {
      "Time": "2022-12-22T15:26:34.3763255+08:00",
      "Group": "Settings",
      "Message": "Presets have been refreshed. Loaded 0 presets.",
      "StackTrace": null,
      "Level": 1,
      "Notification": false
    },
    {
      "Time": "2022-12-22T15:26:34.3763301+08:00",
      "Group": "HttpRequestException",
      "Message": "An attempt was made to access a socket in a way forbidden by its access permissions. (api.github.com:443)",
      "StackTrace": "   at System.Net.Http.HttpConnectionPool.ConnectToTcpHostAsync(String host, Int32 port, HttpRequestMessage initialRequest, Boolean async, CancellationToken cancellationToken)\r\n   at System.Net.Http.HttpConnectionPool.ConnectAsync(HttpRequestMessage request, Boolean async, CancellationToken cancellationToken)\r\n   at System.Net.Http.HttpConnectionPool.CreateHttp11ConnectionAsync(HttpRequestMessage request, Boolean async, CancellationToken cancellationToken)\r\n   at System.Net.Http.HttpConnectionPool.AddHttp11ConnectionAsync(HttpRequestMessage request)\r\n   at System.Threading.Tasks.TaskCompletionSourceWithCancellation`1.WaitWithCancellationAsync(CancellationToken cancellationToken)\r\n   at System.Net.Http.HttpConnectionPool.GetHttp11ConnectionAsync(HttpRequestMessage request, Boolean async, CancellationToken cancellationToken)\r\n   at System.Net.Http.HttpConnectionPool.SendWithVersionDetectionAndRetryAsync(HttpRequestMessage request, Boolean async, Boolean doRequestAuth, CancellationToken cancellationToken)\r\n   at System.Net.Http.DecompressionHandler.SendAsync(HttpRequestMessage request, Boolean async, CancellationToken cancellationToken)\r\n   at System.Net.Http.HttpClient.<SendAsync>g__Core|83_0(HttpRequestMessage request, HttpCompletionOption completionOption, CancellationTokenSource cts, Boolean disposeCts, CancellationTokenSource pendingRequestsCts, CancellationToken originalCancellationToken)\r\n   at Octokit.Internal.HttpClientAdapter.SendAsync(HttpRequestMessage request, CancellationToken cancellationToken) in /home/runner/work/octokit.net/octokit.net/Octokit/Http/HttpClientAdapter.cs:line 220\r\n   at Octokit.Internal.HttpClientAdapter.Send(IRequest request, CancellationToken cancellationToken) in /home/runner/work/octokit.net/octokit.net/Octokit/Http/HttpClientAdapter.cs:line 74\r\n   at Octokit.Connection.RunRequest(IRequest request, CancellationToken cancellationToken) in /home/runner/work/octokit.net/octokit.net/Octokit/Http/Connection.cs:line 675\r\n   at Octokit.Connection.Run[T](IRequest request, CancellationToken cancellationToken) in /home/runner/work/octokit.net/octokit.net/Octokit/Http/Connection.cs:line 666\r\n   at Octokit.ApiConnection.Get[T](Uri uri, IDictionary`2 parameters) in /home/runner/work/octokit.net/octokit.net/Octokit/Http/ApiConnection.cs:line 72\r\n   at OpenTabletDriver.Desktop.Updater.Updater.CheckForUpdates(Boolean forced)",
      "Level": 3,
      "Notification": false
    }
  ]
}

jamesbt365, 2022-12-22:
Fixed in #2391. You may take this configurations file and move it into a folder called `Configurations` next to the executables of otd.

closed 2022-12-22 by jamesbt365

holahapi, 2022-12-25:
> Fixed in #2391. You may take this configurations file and move it into a folder called `Configurations` next to the executables of otd.

thank you~