OpenTabletDriver/OpenTabletDriver #2561: Support Request: cursor teleporting

0120-500-500, 2022-12-28:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### What do you need help with?

opentabletdriver v0.6.0.4
wacom ctl-472

Cursor oscillates up and down or teleports
Since installing this Windows
I have not installed any driver other than OpenTabletDriver

### What have you tried?

Restarted Windows.
I reconnected the tablet
Restarted OpenTabletDriver

### Diagnostic Information

{
  "App Version": "OpenTabletDriver v0.6.0.4",
  "Build Date": "5/4/2022",
  "Operating System": {
    "Platform": 2,
    "ServicePack": "",
    "Version": "10.0.22621.0",
    "VersionString": "Microsoft Windows NT 10.0.22621.0"
  },
  "Environment Variables": {
    "USER": null,
    "TEMP": "C:\\Users\\AL_RE\\AppData\\Local\\Temp",
    "TMP": "C:\\Users\\AL_RE\\AppData\\Local\\Temp",
    "TMPDIR": null,
    "USERPROFILE": "C:\\Users\\AL_RE"
  },
  "HID Devices": [
    {
      "DevicePath": "\\\\?\\hid#vid_046d&pid_c345&mi_02#a&28f5cec5&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "Manufacturer": "Logitech",
      "ProductName": "Logi TKL Mechanical Keyboard",
      "SerialNumber": "",
      "FriendlyName": "Logi TKL Mechanical Keyboard",
      "VendorID": 1133,
      "ProductID": 49989,
      "InputReportLength": 65,
      "OutputReportLength": 65,
      "FeatureReportLength": 9,
      "CanOpen": true
    },
    {
      "DevicePath": "\\\\?\\hid#vid_046d&pid_c539&mi_02&col01#9&2bae006&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "Manufacturer": "Logitech",
      "ProductName": "USB Receiver",
      "SerialNumber": "",
      "FriendlyName": "USB Receiver",
      "VendorID": 1133,
      "ProductID": 50489,
      "InputReportLength": 7,
      "OutputReportLength": 7,
      "FeatureReportLength": 0,
      "CanOpen": true
    },
    {
      "DevicePath": "\\\\?\\hid#vid_046d&pid_c539&mi_02&col03#9&2bae006&0&0002#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "Manufacturer": "Logitech",
[... 1830 lines cut ...]
      "Time": "2022-12-29T01:26:25.155003+09:00",
      "Group": "Settings",
      "Message": "Driver is enabled.",
      "StackTrace": null,
      "Level": 1,
      "Notification": false
    },
    {
      "Time": "2022-12-29T01:26:25.1550078+09:00",
      "Group": "Display",
      "Message": "DPI Awareness enabled",
      "StackTrace": null,
      "Level": 0,
      "Notification": false
    },
    {
      "Time": "2022-12-29T01:26:25.1550126+09:00",
      "Group": "Settings",
      "Message": "Presets have been refreshed. Loaded 0 presets.",
      "StackTrace": null,
      "Level": 1,
      "Notification": false
    }
  ]
}

jamesbt365, 2022-12-28:
As your explanation doesn't really explain it properly this is either chatter or device drivers fucking you over.

- Open "Device Manager"
- At the top click view > device by container
- Look at the list of devices
- Find your tablet
- Open the dropdown menu (container) next to the tablet
- In the drop down container, find any of the following and right click them: <OEM> tablet (e.g. “Wacom Tablet”; “Gaomon Tablet”), the model number (e.g. “CTL-472”), or the name of the tablet (“Wacom Intuos S”)
- Click on update -> browse my computer for drivers -> let me pick from a list of available drivers -> usb input device, usb composite device, or hid compliant device -> next
- Repeat the first two steps if any other matching devices are present
- Replug your tablet and check if otd detects it

closed 2022-12-28 by jamesbt365

0120-500-500, 2022-12-29:
If you follow the above steps and reconnect the tablet
OpenTabletDriver recognizes the tablet
The cursor vibration is still there.

jamesbt365, 2022-12-29:
If its minor vibration back and forth its normal, and a part of physics. And as physics is not something you can control, nothing is broken.

0120-500-500, 2022-12-29:
Vibrates and teleports up and down

jamesbt365, 2022-12-29:
Get a video.

0120-500-500, 2022-12-29:
https://user-images.githubusercontent.com/103758556/209889463-46692508-6bd3-4af5-8287-98a1a8e4816e.mp4

jamesbt365, 2022-12-29:
You have another tablet driver getting in the way.

FlySlime, 2023-09-25:
I also have this issue, and I've ensured that the built-in wacom driver is turned off. It ONLY occurs when I'm using "Artist Mode".

``blacklist wacom`` is included in ``/etc/modprobe.d/blacklist``
```lsmod | grep wacom``` returns nothing.

What's going on?

I'm using GNOME 44 Wayland.

FlySlime, 2023-09-25:
This solved it for me:

              if it is everywhere and not just osu, and file > reset to defaults doesn't fix it then you most probably have another driver installed, the otd console tab should show this.

_Originally posted by @jamesbt365 in https://github.com/OpenTabletDriver/OpenTabletDriver/issues/2234#issuecomment-1120517215_