OpenTabletDriver/OpenTabletDriver #2562: Support Request: Gaomon S620  is not detected

nikololiahim, 2022-12-28:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### What do you need help with?

I need OTD to detect my Gaomon S620 tablet.

### What have you tried?

0. Installed the .NET framework v6 from the link in the installation guide.
1. First, I uninstalled my existing Gaomon driver using the Control Panel and ran the [WinUSBCleanup](https://github.com/X9VoiD/WinUSBCleanup/releases/latest) tool 
2. Downloaded OTD v0.6.4.4 and ran the daemon, then the UI .exe files.
3. Greeted with `[Detect:Info]	No tablets were detected.`
4. Unplugged the tablet, plugged it back in, same story.
5. Restarted the laptop, did step 4 again, no luck.
6. Tried steps 2-5 with all the 0.6.x releases, also no luck.

### Diagnostic Information

Here is the diagnostic dump:
<details>
  <summary>dump.json</summary>

```json
{
  "App Version": "OpenTabletDriver v0.6.0.4",
  "Build Date": "5/4/2022",
  "Operating System": {
    "Platform": 2,
    "ServicePack": "",
    "Version": "10.0.19044.0",
    "VersionString": "Microsoft Windows NT 10.0.19044.0"
  },
  "Environment Variables": {
    "USER": null,
    "TEMP": "C:\\Users\\olomi\\AppData\\Local\\Temp",
    "TMP": "C:\\Users\\olomi\\AppData\\Local\\Temp",
    "TMPDIR": null,
    "USERPROFILE": "C:\\Users\\olomi"
  },
  "HID Devices": [
    {
      "DevicePath": "\\\\?\\usb#vid_5986&pid_116f&mi_04#6&f608b57&1&0004#{dee824ef-729b-4a0e-9c14-b7117d33a817}",
      "Manufacturer": " ",
      "ProductName": "HD Camera",
      "SerialNumber": "0001",
      "FriendlyName": "HD Camera",
      "VendorID": 22918,
      "ProductID": 4463,
      "InputReportLength": 0,
      "OutputReportLength": 0,
      "FeatureReportLength": 0,
      "CanOpen": true
    },
    {
      "DevicePath": "\\\\?\\hid#vid_0457&pid_1595&col05#6&3594eb73&0&0004#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "Manufacturer": "Silicon Integrated System Co.",
      "ProductName": "SiS HID Touch Controller",
      "SerialNumber": "",
      "FriendlyName": "SiS HID Touch Controller",
      "VendorID": 1111,
      "ProductID": 5525,
      "InputReportLength": 6,
      "OutputReportLength": 0,
      "FeatureReportLength": 0,
      "CanOpen": false
[... 2047 lines cut ...]
      "Notification": false
    },
    {
      "Time": "2022-12-28T21:32:35.9036791+03:00",
      "Group": "Display",
      "Message": "DPI Awareness enabled",
      "StackTrace": null,
      "Level": 0,
      "Notification": false
    },
    {
      "Time": "2022-12-28T21:32:35.9036823+03:00",
      "Group": "Settings",
      "Message": "Presets have been refreshed. Loaded 0 presets.",
      "StackTrace": null,
      "Level": 1,
      "Notification": false
    }
  ]
}
```

</details>

Here is the output of the "device string reader" with VendorID 9580 and ProductID 111: [dump.txt](https://drive.google.com/file/d/1jaSiAVQFQ_3XcKjsKNjkbgL0-9d_ZJQ6/view?usp=sharing)

jamesbt365, 2022-12-28:
#2499

closed 2022-12-28 by jamesbt365

nikololiahim, 2022-12-28:
It worked, thank you! The json config in one of the links is missing a curly brace, though. 
Is there any way I can contribute to the docs to make this solution more accessible?