OpenTabletDriver/OpenTabletDriver #2573: scale > full area doesn't scale properly

jamesbt365, 2023-01-04:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Steps to Reproduce

1) Open OpenTabletDriver
2) Make sure lock aspect ratio is enabled
3) right click your tablet area > scale > full
4) area is set incorrectly.

### Expected Behavior

The area should scale to the maximum limits of the tablet and not further.

### Observed Behavior

It scales beyond this.

This likely happens due to different screen aspect ratio, but I'm not too sure and haven't tested further.

https://cdn.discordapp.com/attachments/724106667964366968/1060275495062745119/2023-01-04_19-10-56.mp4

### Tablet Device

PTZ-630

### Diagnostic Information

N/A. Version based off a5d95956785686e31b9a90545f1663c0a50c06b9.

gonX, 2025-01-25:
Does this happen on 0.6.x? If not, this is fixed in Avalonia.

WildPenquin, 2025-03-27:
This is still an issue, if I understand the issue correctly. I can reproduce this on an H641P (Huion Inspiroy S) and a display with 3440x1440 ratio. The tablet is roughly 16:10 and the display 21:9 (actually 21.5:9 or 43:18).

What I'm experiencing is that the area extend horizontally over the tablet area if I follow the steps in the OP.

Moreover, it is not at all clear what is the reference for "lock aspect ratio". One could assume the aspect ratio of movement on the tablet is always 1:1 as on the display. However, one can achieve this by either a) reducing to usable area on the drawing tablet **or** b) on the display. I'd like to do the latter, but it seems impossible.

FWIW I can "work around" the issue by the following steps:

1. Open OpenTabletDriver
2. **Uncheck** Lock Aspect Ratio
3. Click on the input area: Scale -> Full
4. **Check** Lock Aspect Ratio

Results: The area on input is reduced.

What I'd want is the area on the Output to be reduced. This will often be much much more desireable if the output is of a much different aspect ratio than input, and input device is somewhat small, though I can see many users will like to reduce the area on the input in the converse cases (large enough input, aspect ratio is close to output).

This is on Arch Linux, OpenTabletDriver-git (0.7.0.0-pre or v0.7.0.0pre.r1094.gd4150dea-2).

Cheers!

[image]

jamesbt365, 2025-03-27:
>This is still an issue
>This is on Arch Linux, OpenTabletDriver-git (0.7.0.0-pre or v0.7.0.0pre.r1094.gd4150dea-2).

the `master` branch is not actively maintained and is abandoned, as such we do not accept issue reports on it, you should test `0.6.x` (latest release) or `avalonia`.

gonX, 2025-08-31:
I can't seem to reproduce this in 0.6.x, but I do remember `master` having weird GUI. Closing as resolved in `avalonia` and as not a bug present in 0.6.x.

PR #2762 (2023-05-26): Fix some aspect ratio lock issues
- Fixes aspect ratio lock hooking more than once, which prevents it from getting unhooked cleanly
- Resolves #2573 for `0.6.1`