OpenTabletDriver/OpenTabletDriver #2577: Throws errors on launch and will not detect tablets

Axilotl17, 2023-01-06:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Steps to Reproduce

when running with build or release on Arch linux, open the daemon and `OpenTabletDriver.UX.Gtk` (order doesn't matter)

### Expected Behavior

It should have opened normally and let me edit tablet settings.

### Observed Behavior

When they are both open, you get these error messages: 

```
System.Reflection.ReflectionTypeLoadException: Unable to load one or more of the requested types.
Could not load file or assembly 'OpenTabletDriver.Plugin, Version=0.6.0.2, Culture=neutral, PublicKeyToken=null'. The system cannot find the file specified.

   at System.Reflection.RuntimeModule.GetTypes(RuntimeModule module)
   at OpenTabletDriver.Desktop.Reflection.PluginManager.<>c.<get_PluginTypes>b__22_0(Assembly asm) in /home/axilotl/OpenTabletDriver/OpenTabletDriver.Desktop/Reflection/PluginManager.cs:line 58
   at System.Linq.Enumerable.SelectManyIterator[TSource,TCollection,TResult](IEnumerable`1 source, Func`2 collectionSelector, Func`3 resultSelector)+MoveNext()
   at System.Linq.Enumerable.WhereSelectEnumerableIterator`2.MoveNext()
   at System.Linq.Enumerable.WhereEnumerableIterator`1.ToArray()
   at System.Collections.Immutable.ImmutableArray.CreateRange[T](IEnumerable`1 items)
   at OpenTabletDriver.UX.Controls.OutputModePicker.CreateDefaultDataStore() in /home/axilotl/OpenTabletDriver/OpenTabletDriver.UX/Controls/OutputModePicker.cs:line 24
   at OpenTabletDriver.UX.Controls.OutputModePicker.<Refresh>b__3_0() in /home/axilotl/OpenTabletDriver/OpenTabletDriver.UX/Controls/OutputModePicker.cs:line 29
   at Eto.GtkSharp.Forms.ApplicationHandler.<>c__DisplayClass20_0.<AsyncInvoke>b__0(Object <p0>, EventArgs <p1>)
   at Gtk.Application.InvokeCB.Invoke()
   at GLib.Timeout.TimeoutProxy.Handler()
System.IO.FileNotFoundException: Could not load file or assembly 'OpenTabletDriver.Plugin, Version=0.6.0.2, Culture=neutral, PublicKeyToken=null'. The system cannot find the file specified.

File name: 'OpenTabletDriver.Plugin, Version=0.6.0.2, Culture=neutral, PublicKeyToken=null'
```
and then 
```
System.Reflection.ReflectionTypeLoadException: Unable to load one or more of the requested types.
Could not load file or assembly 'OpenTabletDriver.Plugin, Version=0.6.0.2, Culture=neutral, PublicKeyToken=null'. The system cannot find the file specified.

   at System.Reflection.RuntimeModule.GetTypes(RuntimeModule module)
   at OpenTabletDriver.Desktop.Reflection.PluginManager.<>c.<get_PluginTypes>b__22_0(Assembly asm) in /home/axilotl/OpenTabletDriver/OpenTabletDriver.Desktop/Reflection/PluginManager.cs:line 58
   at System.Linq.Enumerable.SelectManyIterator[TSource,TCollection,TResult](IEnumerable`1 source, Func`2 collectionSelector, Func`3 resultSelector)+MoveNext()
   at System.Linq.Enumerable.WhereSelectEnumerableIterator`2.MoveNext()
   at System.Linq.Enumerable.WhereEnumerableIterator`1.ToArray()
   at System.Collections.Immutable.ImmutableArray.CreateRange[T](IEnumerable`1 items)
   at OpenTabletDriver.UX.Controls.OutputModePicker.CreateDefaultDataStore() in /home/axilotl/OpenTabletDriver/OpenTabletDriver.UX/Controls/OutputModePicker.cs:line 24
   at OpenTabletDriver.UX.Controls.OutputModePicker.<Refresh>b__3_0() in /home/axilotl/OpenTabletDriver/OpenTabletDriver.UX/Controls/OutputModePicker.cs:line 29
   at Eto.GtkSharp.Forms.ApplicationHandler.<>c__DisplayClass20_0.<AsyncInvoke>b__0(Object <p0>, EventArgs <p1>)
   at Gtk.Application.InvokeCB.Invoke()
   at GLib.Timeout.TimeoutProxy.Handler()
System.IO.FileNotFoundException: Could not load file or assembly 'OpenTabletDriver.Plugin, Version=0.6.0.2, Culture=neutral, PublicKeyToken=null'. The system cannot find the file specified.

File name: 'OpenTabletDriver.Plugin, Version=0.6.0.2, Culture=neutral, PublicKeyToken=null'
```

you are then presented with the "Connecting to OpenTabletDriver Daemon" screen, and detecting tablets does nothing.

### Tablet Device

Wacom Intuos CTL-4100 (Intuos S)

### Diagnostic Information

[diagnostics.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/10357701/diagnostics.txt)

jamesbt365, 2023-01-06:
you cannot use old plugins on the git version. Either remove them or don't use the git version.

jamesbt365, 2023-01-06:
#2407

closed 2023-01-06 by jamesbt365

Axilotl17, 2023-01-06:
I... didn't realize i had plugins. thank you.

Axilotl17, 2023-01-06:
But even when I used the not-git version it still didn't work for me, FYI.

Axilotl17, 2023-01-06:
And... I can't seem to edit my plugins. when I open plugin manager I get this error: 
```System.InvalidOperationException: No service for type 'OpenTabletDriver.Desktop.Contracts.IDriverDaemon' has been registered.
   at Microsoft.Extensions.DependencyInjection.ServiceProviderServiceExtensions.GetRequiredService(IServiceProvider provider, Type serviceType)
   at OpenTabletDriver.Desktop.DependencyInjectionExtensions.GetServiceForParameter(ParameterInfo parameter, IServiceProvider serviceProvider, IEnumerable`1 deps) in /home/axilotl/OpenTabletDriver/OpenTabletDriver.Desktop/DependencyInjectionExtensions.cs:line 76
   at OpenTabletDriver.Desktop.DependencyInjectionExtensions.<>c__DisplayClass2_0.<CreateInstance>b__3(ParameterInfo p) in /home/axilotl/OpenTabletDriver/OpenTabletDriver.Desktop/DependencyInjectionExtensions.cs:line 50
   at System.Linq.Enumerable.SelectArrayIterator`2.ToArray()
   at OpenTabletDriver.Desktop.DependencyInjectionExtensions.CreateInstance(IServiceProvider serviceProvider, Type type, Object[] additionalDependencies) in /home/axilotl/OpenTabletDriver/OpenTabletDriver.Desktop/DependencyInjectionExtensions.cs:line 49
   at OpenTabletDriver.Desktop.DependencyInjectionExtensions.CreateInstance[T](IServiceProvider serviceProvider, Object[] additionalDependencies) in /home/axilotl/OpenTabletDriver/OpenTabletDriver.Desktop/DependencyInjectionExtensions.cs:line 58
   at OpenTabletDriver.Desktop.DependencyInjectionExtensions.GetOrCreateInstance[T](IServiceProvider serviceProvider, Object[] additionalDependencies) in /home/axilotl/OpenTabletDriver/OpenTabletDriver.Desktop/DependencyInjectionExtensions.cs:line 66
   at OpenTabletDriver.UX.App.ShowWindow[TWindow](Object[] deps) in /home/axilotl/OpenTabletDriver/OpenTabletDriver.UX/App.cs:line 279
   at OpenTabletDriver.UX.App.ShowWindow[TWindow]() in /home/axilotl/OpenTabletDriver/OpenTabletDriver.UX/App.cs:line 269
   at OpenTabletDriver.UX.Components.AppCommand.<>c__DisplayClass6_0.<AddActionHandler>b__0(Object _, EventArgs _) in /home/axilotl/OpenTabletDriver/OpenTabletDriver.UX/Components/AppCommand.cs:line 40
   at Eto.Forms.Command.OnExecuted(EventArgs e)
   at Eto.Forms.Command.Execute()
   at Eto.Forms.Command.System.Windows.Input.ICommand.Execute(Object parameter)
   at Eto.PropertyStore.CommandWrapper.Command_Execute(Object sender, EventArgs e)
   at Eto.Forms.MenuItem.OnClick(EventArgs e)
   at Eto.Forms.MenuItem.Callback.OnClick(MenuItem widget, EventArgs e)
   at Eto.GtkSharp.Forms.Menu.ButtonMenuItemHandler`2.ButtonMenuItemConnector.HandleActivated(Object sender, EventArgs e)
   at InvokeStub_EventHandler.Invoke(Object, Object, IntPtr*)
   at System.Reflection.MethodInvoker.Invoke(Object obj, IntPtr* args, BindingFlags invokeAttr)
   ```

Axilotl17, 2023-01-06:
Moreover, using the normal version after completely reinstalling just times out whilst connecting to daemon.

jamesbt365, 2023-01-06:
> And... I can't seem to edit my plugins. when I open plugin manager I get this error:

you need to remove the plugin folder in this case.

>Moreover, using the normal version after completely reinstalling just times out whilst connecting to daemon.

Downgrading settings is not supported. Remove your ~/.config/OpenTabletDriver directory if you're downgrading versions.