OpenTabletDriver/OpenTabletDriver #2578: No pixmap icon on debian

Sarisan, 2023-01-06:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Steps to Reproduce

1. Install OpenTabletDriver.deb
2. Open applications menu

### Expected Behavior

usr/share/pixmaps/otd.png should be included in deb package

### Observed Behavior

usr/share/pixmaps/otd.png does not exist

### Tablet Device

Wacom CTL-472

### Diagnostic Information

```
➜  Downloads bsdtar -xf OpenTabletDriver.deb data.tar.xz 
➜  Downloads tar -xf data.tar.xz ./usr/share                                   
➜  Downloads cat usr/share/applications/OpenTabletDriver.desktop 
[Desktop Entry]
Version=
Name=OpenTabletDriver
Exec=/usr/bin/opentabletdriver
Icon=/usr/share/pixmaps/otd.png
Terminal=false
Type=Application
StartupNotify=true
Categories=Settings;
➜  Downloads ls usr/share/pixmaps/otd.png
ls: cannot access 'usr/share/pixmaps/otd.png': No such file or directory
➜  Downloads
```

hwsmm, 2023-01-06:
Duplicate of https://github.com/OpenTabletDriver/OpenTabletDriver.Packaging/issues/15

jamesbt365, 2023-01-06:
Its on the readme, but I dont know how to make that more clear.

closed 2023-01-06 by jamesbt365