OpenTabletDriver/OpenTabletDriver #2581: Mouse shaking

cyberbasty, 2023-01-07:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Steps to Reproduce

i just use open driver normally

https://user-images.githubusercontent.com/82098265/211165095-04291495-e437-4167-b951-927d35bf3ec8.mp4

### Expected Behavior

no shaking

### Observed Behavior

my mouse is shaking a lot

### Tablet Device

Wacom ET-0405-U

### Diagnostic Information

N/A

jamesbt365, 2023-01-07:
it is a part of physics. Nothing here is a bug.

jamesbt365, 2023-01-07:
Apply smoothing if you don't like it.

closed 2023-01-07 by jamesbt365

cyberbasty, 2023-01-07:
> Apply smoothing if you don't like it.

i have an other pc wich didn't has the bug
i use the same tablet for both

jamesbt365, 2023-01-07:
Then you have some emi caused by devices close by or have some software interfering. As you didn't provide diagnostic information, i cannot help you further on this.

cyberbasty, 2023-01-07:
diagnostc infos ?

jamesbt365, 2023-01-07:
Yes.

cyberbasty, 2023-01-07:
i cant export becuz it's json

cyberbasty, 2023-01-07:
i cant send becuz it's json

jamesbt365, 2023-01-07:
Rename it to txt or zip it then.

cyberbasty, 2023-01-07:
[aaaaaa.zip](https://github.com/OpenTabletDriver/OpenTabletDriver/files/10367090/aaaaaa.zip)

jamesbt365, 2023-01-07:
not an otd issue then. Refer to what i said above about emi or other software.

cyberbasty, 2023-01-07:
what's an emi

jamesbt365, 2023-01-07:
Electromagnetic Interference.

cyberbasty, 2023-01-07:
Envoyé à partir de Outlook pour Android<https://aka.ms/AAb9ysg>

cyberbasty, 2023-01-07:
[image]

cyberbasty, 2023-01-07:
is there somethig that's emi ?

X9VoiD, 2023-01-07:
@cyberbasty This is not a bug in OpenTabletDriver, the hardware itself is weak and so its raw tracking capability. If you want to smooth out your cursor then use any smoothing filters from plugin manager.