OpenTabletDriver/OpenTabletDriver #2596: [Feature, Linux] Add a basic DBus interface alongside the daemon

orowith2os, 2023-01-10:
### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Description

I've been using OTD since I got my drawing tablet, and it's been pretty solid so far. My only complaint is that there isn't a tray icon (or it doesn't like Tray Icons Reloaded on GNOME) and that can make switching presets on the fly a pain.

I know OTD already provides a daemon, but adding a DBus interface alongside that could be useful as some applications might want to talk to OTD to set some options, such as different presets or modes, and going through its own daemon might not be very app-friendly. Using DBus for these applications/use cases can make it an easier process.

Having a DBus interface would also allow for more integration into the system if some effort is put in by the end user/developers, as the desktop environments can add extensions to easily talk to OTD, such as a GNOME extension (why I'm making this feature request).

I'm not asking for a fully featured interface, but rather just something basic that can be expanded as time goes on. The features I'd request from an interface like this first would be querying and setting presets.

X9VoiD, 2023-01-10:
Would improving the CLI not suffice? Platform-specific features (that are not related to input processing) are mostly not welcome here.

wsippel, 2023-01-13:
D-Bus originated on Linux, but it runs on all POSIX operating systems, as well as Windows: https://github.com/freedesktop/dbus/blob/master/README.win
So if a multi-platform application wanted to talk to OTD directly, D-Bus should work on all platforms OTD runs on.

X9VoiD, 2023-01-13:
I still don't believe it offers any real advantage over just improving and making use of the CLI. Adding D-Bus support as an official feature of OTD will just increase maintenance burden on top of being redundant. It would mean the daemon would now have to serve two different kinds of IPC. And that's a nightmare nobody would like to go through without any compelling argument of why it's a *need* to include D-Bus support.

X9VoiD, 2023-01-13:
I'd argue that letting the daemon spit out either a pretty or json formatted output is a better place to put efforts in.

wsippel, 2023-01-13:
> I still don't believe it offers any real advantage over just improving and making use of the CLI. Adding D-Bus support as an official feature of OTD will just increase maintenance burden on top of [...]

Yeah, it would probably only really make sense as the only IPC. So remove whatever else you use, and only use D-Bus for everything. That might be a worthwhile change considering its flexibility and cross-platform nature, depending on how complicated the migration would be. But I totally get not wanting to support two different, independent IPC mechanisms. That sounds like a massive headache. But D-Bus is really simple from what I understand. It just uses pipes, and you could use JSON, TOML, YAML, CSV or whatever you want for the actual communication I believe.

X9VoiD, 2023-01-13:
> So remove whatever else you use, and only use D-Bus for everything

No

wsippel, 2023-01-13:
> > So remove whatever else you use, and only use D-Bus for everything
> 
> No

I mean, I don't personally have a horse in this race, but why? I don't get the feeling you actually worked with D-Bus, maybe it could make things easier in the long run? Not trying to be an ass or anything, but having to maintain an IPC mechanism is work, and here's one that's ready-to-use, that would work everywhere, and potentially open up new possibilities. Maybe that's worth looking into? 

Again, don't get me wrong, I love OTD as is, but I do think a standard mechanism for other applications to interface with OTD could be kinda useful, and D-Bus might be the easiest way to do it, both for you and application developers, but I'm by no means an expert.

X9VoiD, 2023-01-13:
The CLI already exists, and all supported platforms can use it in the exact same way without having to deal with D-Bus.