OpenTabletDriver/OpenTabletDriver #2598: High CPU usage after being open for a while

joojoooo, 2023-01-11:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Steps to Reproduce

No idea, happens randomly, just open the program and use it for a few hours.
Then open Task Manager and you will see the CPU usage way higher than when you first started it.
[image]
Using no plugins, no smoothing, nothing.

### Expected Behavior

I expect it to have the same CPU usage as when it is initially started since nothing changed meanwhile.
I shouldn't be needing to restart it every hour so it doesn't make my game lag.

### Observed Behavior

Does't happen with other drivers like the hawku TabletDriver, I can keep it open forever it will always be at 0% usage.
OpenTabletDriver instead will start at 5%, sometimes 15% CPU usage when I initially open it and will randomly creep up to 30% for no reason.

### Tablet Device

Wacom CTL-480

### Diagnostic Information

```json
{
  "App Version": "OpenTabletDriver v0.6.0.4",
  "Build Date": "5/4/2022",
  "Operating System": {
    "Platform": 2,
    "ServicePack": "",
    "Version": "10.0.19043.0",
    "VersionString": "Microsoft Windows NT 10.0.19043.0"
  },
  "Environment Variables": {
    "USER": null,
    "TEMP": "C:\\Users\\myuser\\AppData\\Local\\Temp",
    "TMP": "C:\\Users\\myuser\\AppData\\Local\\Temp",
    "TMPDIR": null,
    "USERPROFILE": "C:\\Users\\myuser"
  },
  "HID Devices redacted": [
    {
      "DevicePath": "\\\\?\\hid#vid_056a&pid_030e&mi_01#8&294640ab&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "Manufacturer": "Wacom Co.,Ltd.",
      "ProductName": "Intuos PS",
      "SerialNumber": "",
      "FriendlyName": "Intuos PS",
      "VendorID": 1386,
      "ProductID": 782,
      "InputReportLength": 64,
      "OutputReportLength": 0,
      "FeatureReportLength": 0,
      "CanOpen": true
    },
  ],
  "Console Log": [
    {
      "Time": "2023-01-11T05:02:51.2939092+01:00",
      "Group": "Plugin",
      "Message": "Loading plugin 'BezierInterpolator'",
      "StackTrace": null,
      "Level": 0,
      "Notification": false
[... 1707 lines cut ...]
      "Group": "Plugin",
      "Message": "Loading plugin 'HawkuFilters'",
      "StackTrace": null,
      "Level": 0,
      "Notification": false
    },
    {
      "Time": "2023-01-11T05:02:51.2955267+01:00",
      "Group": "Display",
      "Message": "DPI Awareness enabled",
      "StackTrace": null,
      "Level": 0,
      "Notification": false
    },
    {
      "Time": "2023-01-11T05:02:51.2955339+01:00",
      "Group": "Settings",
      "Message": "Presets have been refreshed. Loaded 0 presets.",
      "StackTrace": null,
      "Level": 1,
      "Notification": false
    }
  ]
}
```

joojoooo, 2023-01-11:
Just noticed it says `Unsupported interval detected, will use fallback timer. Expect high CPU usage` in the diagnostic console messages. What is that and why hawku devocub TabletDriver has no such an issue? I really want to switch to OpenTabletDriver due to all the features plugins bring, but at this point it's simply unusable for me. Also, you can't move on administrator windows like task manager unless you run it with administrator privileges but then it gives a warning everytime you do that, which is annoying. hawku driver doesn't even run with administrator and somehow works on task manager.

X9VoiD, 2023-01-11:
Use a `Frequency` that is supported by Windows, 1000hz, 500hz, 250hz, 200hz, 125hz, 100hz, anything as long as the result of the expression `1000 / N` has no fractional part. If you want to let OTD inject input to privileged apps then use VMultiMode plugin. Visit it's wiki for installation details.