OpenTabletDriver/OpenTabletDriver #2607: [WACOM CHT-480] cursor moving when not in relative mode

kiuKisas, 2023-01-14:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Steps to Reproduce

Context: multi-screens and and ultrawide, so I want to set an area according to my tablet dimension
1. launch otd from commandline
2. launch otd UI with this settings:
[image]

### Expected Behavior

The cursor is moving quickly, as if  it altern between my otd setting and the default position.

### Observed Behavior

When I disable otd or set in `relative mode, the cursors is stable. 

### Tablet Device

Wacom CTH-480

### Diagnostic Information

[wacom cth-480.json.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/10417952/wacom.cth-480.json.txt)

jamesbt365, 2023-01-14:
```   {
      "Time": "2023-01-14T15:43:29.0233598+01:00",
      "Group": "Detect",
      "Message": "Another tablet driver found: Wacom",
      "StackTrace": null,
      "Level": 2,
      "Notification": false
    },
    {
      "Time": "2023-01-14T15:43:29.0233824+01:00",
      "Group": "Detect",
      "Message": "Detected input coming from Wacom driver",
      "StackTrace": null,
      "Level": 3,
      "Notification": false
    },
```

You have the wacom kernel module loaded. you can remove it with `rmmod wacom`. the persistant solution is on the wiki.

closed 2023-01-14 by jamesbt365

kiuKisas, 2023-01-14:
Sorry about that, and thank's a lot for your quick answer !