OpenTabletDriver/OpenTabletDriver #2608: Dynamic tip binding

Willie2020, 2023-01-14:
### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Description

I would like to know if there will be a feature for dynamic tip binding whereby when you click on the pen button, the tip changes from to right-click, left-click, and middle-click like the Wacom Click & Tap feature.

jamesbt365, 2023-01-14:
This will likely never be a feature in the base driver but preset changing on a binding is in the upcoming version and this could be done with a plugin.

Willie2020, 2023-01-14:
Alright, thanks. Please can you point me in the right direction regarding the plugin or the upcoming version to follow?

jamesbt365, 2023-01-14:
>Please can you point me in the right direction regarding the plugin

it does not exist, hence why i said it could, not it is.

>or the upcoming version to follow?

Click actions on the top of the github repository. You will not have proper plugin support, so no pressure sensitivity on windows.

Willie2020, 2023-01-14:
oh ok, got it

closed 2023-01-14 by Willie2020