OpenTabletDriver/OpenTabletDriver #2609: High CPU usage after minimizing on "output" tab

BK8000L, 2023-01-15:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Steps to Reproduce

case 1:
start otd and minimize it

case 2:
start otd, select "filters" tab and minimize it

### Expected Behavior

0% cpu usage in a minimized app that does nothing

### Observed Behavior

case 1:
3% of cpu usage

case 2:
0% of cpu usage

### Tablet Device

ctl-470

### Diagnostic Information

[jason.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/10419469/jason.txt)

BK8000L, 2023-01-15:
only process hacker can detect such cpu usage (calculates it differently)
https://processhacker.sourceforge.io/nightly.php

BK8000L, 2023-01-19:
Occurs only with "run: minimized" in shortcut properties

InfinityGhost, 2023-03-14:
@BK8000L Give this [CI build](https://github.com/OpenTabletDriver/OpenTabletDriver/actions/runs/4410789924) a shot, this should fix it.

jamesbt365, 2023-03-15:
its not our issue, please do not cross post.

BK8000L, 2023-03-15:
> please do not cross post

what do you expect since you locked that issue?

> its not our issue

no driver: works fine
wacom driver: works fine
devocub/hawku: works fine
otd: doesn't work properly

InfinityGhost, 2023-03-15:
Again, that issue is not something within our control and out of scope of this issue. See https://github.com/OpenTabletDriver/OpenTabletDriver/issues/2619#issuecomment-1403804188

PR #2659 (2023-03-15): Switch AreaDisplay to standard Drawable
- May fix #2637
- Fixes #2609
closed 2023-03-15 by X9VoiD