OpenTabletDriver/OpenTabletDriver #2612: OpenTabletDriver cannot detect Huion RTM-500.

KaikuraF, 2023-01-18:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Steps to Reproduce

1.Opening OpenTabletDriver.
2.Trying to find my tablet.
3.Tablet not detected.

### Expected Behavior

I expected opentabletdriver to detect my tablet.

### Observed Behavior

OpenTabletDriver didn’t detect my tablet, despite the fact that it is listed as supported.

### Tablet Device

Huion RTM-500

### Diagnostic Information

[Diagnostics.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/10451275/Diagnostics.txt)

jamesbt365, 2023-01-18:
Not supported in the latest version.

jamesbt365, 2023-01-18:
You can take the configuration from #2282 and put it in a folder called `Configurations` next to the executables of otd.

closed 2023-01-18 by jamesbt365

KaikuraF, 2023-01-18:
Thank you so much!

jamesbt365, 2023-01-19:
Export a diagnostic in the help menu at the top of otd.

KaikuraF, 2023-01-19:
[Diagnostic1.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/10452292/Diagnostic1.txt)

jamesbt365, 2023-01-19:
Plug your tablet in.

KaikuraF, 2023-01-19:
I connected the tablet, however, an identical error occurs.
[Diagnostic2.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/10452322/Diagnostic2.txt)

jamesbt365, 2023-01-19:
Tablet > device string reader
9580
100
201
dump all.

KaikuraF, 2023-01-19:
[Dump.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/10452376/Dump.txt)

KaikuraF, 2023-01-19:
Oh my God! Thank you very much.
It finally worked!