OpenTabletDriver/OpenTabletDriver #2617: Add Brush/Eraser and E/B modes as a Express Key binding

gelatinbomb, 2023-01-24:
### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Description

In the XP-Pen aplication(maybe this is in other graphic tablets but I'm not sure) you have the capacity of asign this modes to any express key on your tablet, is mainly used on the stylus buttons.

The Brush/Eraser mode is a key binding that allows you to dynamically change between your brush and your eraser by just clicking one button. 

I don't know how it works because programs tend to have different shorcuts to those tools so either the driver have a hardcode list of the shorcuts of different software or there is some kind of implementation on the different software that allows it to understand that type of signal. I know XP-Pen offers an sdk so maybe that's the reason why it works? Kind of messy but the second option is way more easy.

The e/b mode is quite straightforward. You select what keystroke is going to be the eraser and what is going to be the brush. Then you establish one by default, in this case the brush mode have more sense. Lastly each time you press the express key the key associated with the brush changes to the key asociated with the eraser and viceversa.

X9VoiD, 2023-01-24:
This already exists on Windows Ink plugin. Did you mean for Artist Mode for linux platforms?

gelatinbomb, 2023-01-25:
Yeah I would like to see it on Linux on the Artist Mode! The XP Pen driver seems to be able to do it on Linux so I guess the possibility is there.
I will asume the Windows Ink plugin does not work on Linux

gonX, 2023-02-03:
Just as a side note, we already support eraser mode if your tablet pen has an eraser end.

I understand you're asking for this toggle to be explicit, however I don't see a good way to expose this to the user that doesn't imply normal eraser usage isn't automatically supported. Would it not be preferable to have this as a keybinding in your application instead?

With that being said, adding support for this _should_ be moderately easy, however there are a couple of pitfalls to note if implemented naively.

Code wise in Linux' Artist Mode, it currently updates eraser state when the tool status changes, such as an `OutOfRangeReport` report or `IEraserReport`, _and_ the tablet parser is configured to emit these reports - meaning, if the user were to enable this on a tablet which already has eraser (or pen proximity out, aka `OutOfRangeReport`) support, if implemented naively (e.g. simply enabling the internal `eraser` state) it would "mysteriously" reset when the tool leaves the area, which is potentially not desirable. Therefore, we have a couple of options for states:
- Auto/clear (like now, but would set eraser state to false if it was on any other option)
- Until reset (pen leaves _and_ requires tablet support _and_ requires that selected parser supports it)
- Force on (always set eraser state)
- Force off (never set eraser state)

I have no idea how XP-Pen implements the option, but I would imagine its an always-on or always- off kinda ordeal, and potentially only available for tablets with no eraser pen support. We could likely hide it for tablets with `IEraserReport` support enabled and then simply force-set the internal eraser state so that it doesn't reset on `OutOfRangeReport`, which would simplify the user experience. AFAIK, we currently don't have a way to only show relevant bindings, so code for this would need to be added as well.

Because of the reasons stated in the previous paragraph, this is sadly going to get put on the backlog.