OpenTabletDriver/OpenTabletDriver #2618: Fedora 37 Gnome Wayland No Input Events Registered

Nyeksenn, 2023-01-25:
### Acknowledgements

- [ ] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Steps to Reproduce

1. Install Fedora 37 with Gnome
2. Use Gnome Wayland session
3. Install RPM package from this [comment](https://github.com/OpenTabletDriver/OpenTabletDriver/issues/1986#issuecomment-1133448072)
4. Start service and trigger udevadm

### Expected Behavior

The service should start and detect the tablet. Afterwards it should be configurable with the GUI.
As a result, the tablet should now be usable.

### Observed Behavior

What acutally happens is that the service start fine. It also detects the tablet. The GUI shows no issue at all.
The device can also be configured using the GUI. If the output of `sudo libinput list-devices` is checked,
the tablet correctly appears. However, no application is registering input from the stylus and `sudo libinput debug-events`
does not show any event triggered by the tablet. I was able to get it to work sometimes if I first booted in a Windows
installation and configured the tablet there with the official driver. I am however not able to reproduce this
consistently.

### Tablet Device

XP-PEN Innovator 16

### Diagnostic Information

- The diagnostics file exported from the GUI [innovator16-diagnostics.json](https://github.com/OpenTabletDriver/OpenTabletDriver/files/10498489/innovator16-diagnostics.txt)
- libinput get-devices output
```bash
Device:           OpenTabletDriver Virtual Artist Tablet
Kernel:           /dev/input/event29
Group:            7
Seat:             seat0, default
Size:             26x14mm
Capabilities:     tablet 
Tap-to-click:     n/a
Tap-and-drag:     n/a
Tap drag lock:    n/a
Left-handed:      n/a
Nat.scrolling:    n/a
Middle emulation: n/a
Calibration:      n/a
Scroll methods:   none
Click methods:    none
Disable-w-typing: n/a
Disable-w-trackpointing: n/a
Accel profiles:   none
Rotation:         n/a
```

Theraa0, 2023-02-04:
I'm getting the same problem on my Arch Linux install with a Huion H320M.

gonX, 2023-02-04:
@Nyeksenn  The uploaded diagnostics file does not seem to indicate any issues. Are there any kernel logs associated with OpenTabletDriver setting up the device? (check `dmesg`)
I am not entirely sure what could be the issue here, other than maybe security stuff like SELinux, which nothing from your logs indicates it is.

Does `evtest <virtual device>` report any events? e.g. your `libinput list-devices` command lists the virtual tablet as `/dev/input/event29`, so that'd be `evtest /dev/input/event29`, but keep in mind the endpoint may change on restart (or reconfiguration).

Additionally, when the device is not operating correctly, can you confirm that the tablet is actually sending packets through the Tablet Debugger in OpenTabletDriver?

gonX, 2023-02-04:
@Theraa0 please make a new issue, as it is another Distro it is likely the issue is different

Nyeksenn, 2023-02-12:
So I checked it again and I stumbled upon some error messages.

First, the Application itself has thrown an error message 
[image]
[image]

I also found an error in the kernel log:
[kernel_log.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/10715852/kernel_log.txt)

Furthermore, the GUI reported this warning:

`[Display:Warning]	Neither Wayland nor X11 were detected, defaulting to X11.`

I also tried using the builtin debugger but I am not sure whether I am using it right.
It just shows a blank output.
[image]

Lastly, the `evtest` command didn't report any event.

jamesbt365, 2023-02-12:
>[Display:Warning]	Neither Wayland nor X11 were detected, defaulting to X11.

For some reason you don't have DISPLAY or WAYLAND_DISPLAY set. This is a problem within your environment.

Nyeksenn, 2023-02-12:
So I rebooted and now the variables are set. Not sure what the problem was.
```bash
$ echo $DISPLAY
$ :0
$ echo $WAYLAND_DISPLAY
$ wayland-0
```
The behavior is still the same. I can't see any input events.
And there is still a segfault in the log
`.NET ThreadPool[2508]: segfault at e0 ip 00007ff8e94e10b4 sp 00007ff8ed5fd0c8 error 4 in libX11.so.6.4.0[7ff8e94cc000+8f000] likely on CPU 13 (core 5, socket 0)`

gonX, 2023-11-21:
If the tablet debugger remains empty this is probably more an issue with the specific initialization routines used for your tablet. Does it work with the stock drivers, or with OpenTabletDriver on Windows?

Nyeksenn, 2023-11-21:
It does work with the official Linux driver
and it also works with opentabletdriver under Windows

gonX, 2023-11-21:
Does it still happen with 0.6.3.0? I just realized you're still on 0.6.0.4.

Nyeksenn, 2023-11-21:
I actually switched to a different device so I haven't checked this issue in a while but I will check.

Chris1320, 2024-06-07:
I am also having the same issue in Fedora Workstation 40 using GNOME 46. My tablet is a Huion 420.

jamesbt365, 2024-06-07:
Yours is likely actually #2229.