OpenTabletDriver/OpenTabletDriver #2619: Horrible polling rate with chrome window behind osu

BK8000L, 2023-01-25:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Steps to Reproduce

use otd on "absulute mode" with devocub filters at 1000hz, open any web page with animated elements in chrome (https://codepen.io/metring/pen/gyLjBE/ or osu MP progress webpage, there is an animated circle), start osu in fullscreen and leave maximized chrome window behind osu

### Expected Behavior

997-1002hz polling rate and smooth cursor movement
[image]

### Observed Behavior

950-1050hz polling rate and jerky cursor movement
[image]

### Tablet Device

any

### Diagnostic Information

[jason.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/10499592/jason.txt)
"Vmulti absolute mode" fixes the issue

jamesbt365, 2023-01-25:
Windows 11?

BK8000L, 2023-01-25:
windows 10 1809

X9VoiD, 2023-01-25:
This is already fixed in master by using higher process priority.

closed 2023-01-25 by X9VoiD

BK8000L, 2023-01-25:
i set daemon to high priority by myself and its still the same, this only happens in fullscreen osu

BK8000L, 2023-01-25:
priority doesn't matter even when stress testing the CPU its the same, minimized chrome - perfect polling, maximized - horrible polling.

BK8000L, 2023-01-25:
i installed win 11 22h2 and issue is also there.
Sometimes polling is fine, but it breaks after minimizing and maximizing chrome

BK8000L, 2023-01-25:
@X9VoiD reopen the issue

X9VoiD, 2023-01-25:
It's something we can't fix from our side then. Just don't have anything in the background consuming extra resources.

BK8000L, 2023-01-25:
@X9VoiD Please don't miss my posts, read them again if you forgot. https://github.com/OpenTabletDriver/OpenTabletDriver/issues/2619#issuecomment-1403647444

BK8000L, 2023-01-25:
my cpu is not loaded more than 35% when this happens

X9VoiD, 2023-01-25:
OTD has no control over how Windows prioritizes what processes to run and it *cannot* force Windows to forcibly run the timers exactly when we need it. There is no silver bullet here to help you. If using VMulti helps your situation then use that (it's the exact same way Hawku/DevocubTD sends input). Locking this.