OpenTabletDriver/OpenTabletDriver #2621: Auxiliary Buttons not available on Huion 420

Maxwelldoug, 2023-01-25:
Hey! Finally got my tablet today, and installed the manufacturer drivers. Buttons worked, but I couldn't move the cursor and click at the same time. This was quite annoying, so I've switched to OTD, which has resolved the issue, but also doesn't seem to have an option to rebind the auxiliary buttons. If someone could help me get them showing (or direct me to where I've missed them, if so) I'd greatly appreciate it.

jamesbt365, 2023-01-25:
The 420 doesn't have buttons. The h420 does.

closed 2023-01-25 by jamesbt365

Maxwelldoug, 2023-01-25:
> The 420 doesn't have buttons. The h420 does.

The 3 physical buttons on my tablet disagree? It loads both in official drivers and OTD as a 420, but I very much do have buttons to press.

jamesbt365, 2023-01-25:
Huion try to put the right firmware on the right tablet challenge (impossible).

Move this file into a folder called `Configurations` next to otd.
https://cdn.discordapp.com/attachments/628651971267788800/1067901546773364826/420.json

Maxwelldoug, 2023-01-25:
Alright, thanks. That is a really stupid issue to have, and I hope that whoever is responsible gets their sh*t together.

o7

jamesbt365, 2023-01-25:
You think huion cares enough to make sure that tablets get the right firmware? The h420 and 420 are almost identical. You can solder buttons onto the 420 and it works fine. They are the same pcb and everything. Just one has buttons.

jamesbt365, 2023-01-25:
Anyway np

Maxwelldoug, 2023-01-25:
that is somehow worse.