OpenTabletDriver/OpenTabletDriver #2626: Auxillary bindings not giving desired output.

Silenstryke, 2023-01-29:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Steps to Reproduce

1. Open opentabletdriver and go to auxiliary settings tab.
2. Set any binding 1->4 to "-" [minus], save and apply.

### Expected Behavior

Pressing the auxiliary button should result in [minus] key press.

### Observed Behavior

Pressing the auxiliary button results in [subtract] key press.

### Tablet Device

CTL-4100

### Diagnostic Information

[diag.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/10530338/diag.txt)

jamesbt365, 2023-01-29:
#2522

Silenstryke, 2023-01-29:
wow i searched and that never came up, should have tried harder i guess. sorry for duplicate.

closed 2025-12-05 by gonX