OpenTabletDriver/OpenTabletDriver #2627: XP-Pen G640S V2 incorrectly selects XP_PenTabletOverflowReport parsing branch and has incorrect position as a result

gonX, 2023-01-31:
[Discord convo](https://discord.com/channels/615607687467761684/723399565780189234/1070024256974503936)

Tablet reports 12 bytes from tablet debugger, which results in `XP_PenTabletOverflowReport` being chosen:
https://github.com/OpenTabletDriver/OpenTabletDriver/blob/961578a267083da98f94423f1d20d3b6956619fc/OpenTabletDriver.Configurations/Parsers/XP_Pen/XP_PenReportParser.cs#L7-L18

On 0.5.3 the tablet was using the default TabletParser and as such did not have any issues:
https://github.com/OpenTabletDriver/OpenTabletDriver/blob/dad5382a44c3514547654d8f295cf3651a971aa7/OpenTabletDriver/Configurations/XP-Pen/Star%20G430S.json#L20 (yes, 0.5.3 did not have a V2 entry)

Seems to be a regression caused by #1420, since the tablet was already using the report parser at this point @X9VoiD
https://github.com/OpenTabletDriver/OpenTabletDriver/blob/30b996c90c43575ed1c3ecaa7c5c692dd232be15/OpenTabletDriver.Configurations/Configurations/XP-Pen/Star%20G430S%20V2.json#L26

gonX, 2023-01-31:
Worth noting that the specific tablet this was occurring to might have funky firmware, since the user says its usable on Windows (OTD version unknown, but assumed 0.5.3 since its "2 years old"), but the tablet recording (attached) 
[tablet-data(2).txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/10549425/tablet-data.2.txt)
is really messy and makes no sense to me.

X9VoiD, 2023-01-31:
The symptoms that the user is having is not related to the parser code update, nothing is reported on the last bytes and wouldn't have affected the values.

0 shifted to the right is 0. Anything OR'ed with zero produces the same value.

closed 2023-01-31 by X9VoiD