OpenTabletDriver/OpenTabletDriver #2636: Support Request:

peltsippi, 2023-02-10:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### What do you need help with?

Everythign else works perfectly but eraser is not detected.

### What have you tried?

Restart computer, switch assignment of eraser to different keys, check tablet debugger data etc. Tried million ways to find and remove old wacom drivers.

This is what I see when using the eraser side of the pen:
[image]

### Diagnostic Information

´´´json

{
  "App Version": "OpenTabletDriver v0.6.0.4",
  "Build Date": "5/4/2022",
  "Operating System": {
    "Platform": 2,
    "ServicePack": "",
    "Version": "10.0.22621.0",
    "VersionString": "Microsoft Windows NT 10.0.22621.0"
  },
  "Environment Variables": {
    "USER": null,
    "TEMP": "C:\\Users\\timop\\AppData\\Local\\Temp",
    "TMP": "C:\\Users\\timop\\AppData\\Local\\Temp",
    "TMPDIR": null,
    "USERPROFILE": "C:\\Users\\timop"
  },
  "HID Devices": [
    {
      "DevicePath": "\\\\?\\usb#vid_0b05&pid_1939&mi_00#7&40fe908&0&0000#{dee824ef-729b-4a0e-9c14-b7117d33a817}",
      "Manufacturer": "AsusTek Computer Inc.",
      "ProductName": "AURA LED Controller",
      "SerialNumber": "9876543210",
      "FriendlyName": "AURA LED Controller",
      "VendorID": 2821,
      "ProductID": 6457,
      "InputReportLength": 0,
      "OutputReportLength": 0,
      "FeatureReportLength": 0,
      "CanOpen": true
    },
    {
      "DevicePath": "\\\\?\\hid#vid_046d&pid_c539&mi_02&col02#a&1938b894&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "Manufacturer": "Logitech",
      "ProductName": "USB Receiver",
      "SerialNumber": "",
      "FriendlyName": "USB Receiver",
      "VendorID": 1133,
      "ProductID": 50489,
      "InputReportLength": 20,
      "OutputReportLength": 20,
      "FeatureReportLength": 0,
      "CanOpen": true
    },
    {
      "DevicePath": "\\\\?\\hid#vid_046d&pid_c539&mi_02&col01#a&1938b894&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "Manufacturer": "Logitech",
      "ProductName": "USB Receiver",
      "SerialNumber": "",
[... 2090 lines cut ...]
      "Message": "Driver is enabled.",
      "StackTrace": null,
      "Level": 1,
      "Notification": false
    },
    {
      "Time": "2023-02-10T22:40:00.0362159+02:00",
      "Group": "Tablet",
      "Message": "Tablet debugging is enabled",
      "StackTrace": null,
      "Level": 0,
      "Notification": false
    },
    {
      "Time": "2023-02-10T22:40:00.0362201+02:00",
      "Group": "Tablet",
      "Message": "Tablet debugging is disabled",
      "StackTrace": null,
      "Level": 0,
      "Notification": false
    }
  ]
}

```

jamesbt365, 2023-08-23:
I do see eraser support in the parser this tablet uses (at least now)

Could you get data recordings with the eraser please? (tablet > tablet debugger)

peltsippi, 2023-09-07:
[tablet-data.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/12551992/tablet-data.txt)

Here is output. First with pen tip and then pen tip while button 1 pressed, then pen tip with button 2 pressed. After that eraser and pen alternating.

gonX, 2023-10-01:
How are you expecting it to work exactly? I see the eraser is getting detected fine in your latest `tablet-data.txt`

Keep in mind that Wacom tablets tend to send eraser reports with tool entry (e.g. line 2467 in your `tablet-data.txt`), so we are not actively reporting eraser end on every subsequent report. 

This is supposed to be fixed by #2557 but I will find my GD-0608-U (which uses the same parser) and test eraser behavior.

gonX, 2023-10-01:
Actually, I just realized you are using 0.6.0.4. This was not fixed before v0.6.1, which released after this issue. Closing as solved.

closed 2023-10-01 by gonX