OpenTabletDriver/OpenTabletDriver #2637: Take all of my RAM on Pop! OS

LucasStf, 2023-02-13:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Steps to Reproduce

1: Open the application
2: Minimize the application and see that the ram goes up
3: Then after some times, the application take all of my 32Go of ram.
[open_tablet_driver_ram.webm](https://user-images.githubusercontent.com/97894103/218581673-74799d99-8001-44e3-babf-d8583acbedf9.webm)

### Expected Behavior

The ram usage should be normal

### Observed Behavior

The ram usage goes up with time to fully take all of my 32go of ram

### Tablet Device

Wacom CTL 472

### Diagnostic Information

[file.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/10726597/file.txt)

Inconn, 2023-02-19:
can't replicate on arch linux w/ opentabletdriver aur package on wayland.

jamesbt365, 2023-08-23:
I'm aware that the UX in 0.6.x still has some memory leaks, as well as poor memory management with logging, but should be a separate issue tracking the latter. The former should be resolved in #2772.

gonX, 2023-10-03:
Worth noting that OP uses 2 plugins:
- HawkuFilters Smoothing
- Kuuube's CHATTER EXTERMINATOR Raw

Issue is with 0.6.0.4 - does this still happen in 0.6.3.0?

jamesbt365, 2025-01-24:
Closing as stale, please reopen if large memory leaks can still be reproduced.

PR #2659 (2023-03-15): Switch AreaDisplay to standard Drawable
- May fix #2637
- Fixes #2609
closed 2025-01-24 by jamesbt365