OpenTabletDriver/OpenTabletDriver #2638: Buttons not working in Linux Artist mode (working in relative)

Belamat, 2023-02-15:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Steps to Reproduce

set open tablet driver with Artist mode and set binding buttons to Middle and Right buttons

### Expected Behavior

to function as normal 

### Observed Behavior

buttons didnt work

### Tablet Device

xp pen ct430 deco fun

### Diagnostic Information

[diagnostics.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/10746302/diagnostics.txt)

jamesbt365, 2023-02-15:
You need to use artist mode buttons in artist mode.

closed 2023-02-15 by jamesbt365

Belamat, 2023-02-15:
> You need to use artist mode buttons in artist mode.

Hi James, I tried that, putting artist mode buttons but then nothing anything else should be done to get them to work? thanks