OpenTabletDriver/OpenTabletDriver #2639: PTH-460 works on Windows, but not on M1 Mac

chungis-soup, 2023-02-17:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Steps to Reproduce

Open tablet driver doesn't recognize the PTH-460 after plugging it in on an M1 Mac.  I have tried cmd+D and plug/replugging multiple times but it still doesn't recognize it.  My Ctl-480 works without issues so all the permissions on the Mac are met.

### Expected Behavior

The tablet is on the supported tablet list and works on Windows. 

### Observed Behavior

On the console, it says Failed to open USB device: 0xE00002C5.

### Tablet Device

PTH-460

### Diagnostic Information

[tabletDiag.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/10769311/tabletDiag.txt)

jamesbt365, 2023-02-17:
From a quick google search this seems like another driver/application is grabbing the device before otd can.

gonX, 2023-11-19:
This seems to be a duplicate of #2497 actually

closed 2023-11-19 by gonX