OpenTabletDriver/OpenTabletDriver #2642: Huion H640P Not detected

Clankx, 2023-02-21:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### What do you need help with?

Opentabletdriver cant detect my tablet, Huion H640P.

### What have you tried?

I have uninstalled the huion driver, ran tabletdriver cleanup and restarted, it still doesnt detect it.

### Diagnostic Information

{
  "App Version": "OpenTabletDriver v0.6.0.4",
  "Build Date": "5/4/2022",
  "Operating System": {
    "Platform": 2,
    "ServicePack": "",
    "Version": "10.0.19045.0",
    "VersionString": "Microsoft Windows NT 10.0.19045.0"
  },
  "Environment Variables": {
    "USER": null,
    "TEMP": "C:\\Users\\Lorenzo\\AppData\\Local\\Temp",
    "TMP": "C:\\Users\\Lorenzo\\AppData\\Local\\Temp",
    "TMPDIR": null,
    "USERPROFILE": "C:\\Users\\Lorenzo"
  },
  "HID Devices": [
    {
      "DevicePath": "\\\\?\\hid#vid_256c&pid_0064&mi_02#7&17a3a33b&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}\\kbd",
      "Manufacturer": "HUION",
      "ProductName": "Huion Tablet_H640P",
      "SerialNumber": "",
      "FriendlyName": "Huion Tablet_H640P",
      "VendorID": 9580,
      "ProductID": 100,
      "InputReportLength": 8,
      "OutputReportLength": 0,
      "FeatureReportLength": 0,
      "CanOpen": false
    },
    {
      "DevicePath": "\\\\?\\hid#vid_256c&pid_0064&mi_01#7&c339a87&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "Manufacturer": "HUION",
      "ProductName": "Huion Tablet_H640P",
      "SerialNumber": "",
      "FriendlyName": "Huion Tablet_H640P",
      "VendorID": 9580,
      "ProductID": 100,
      "InputReportLength": 10,
      "OutputReportLength": 0,
      "FeatureReportLength": 0,
      "CanOpen": false
    },
    {
      "DevicePath": "\\\\?\\hid#vid_248a&pid_8367&mi_00&col01#7&238f0916&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "Manufacturer": "Telink",
      "ProductName": "Wireless Receiver",
      "SerialNumber": "",
      "FriendlyName": "Wireless Receiver",
      "VendorID": 9354,
      "ProductID": 33639,
      "InputReportLength": 5,
      "OutputReportLength": 0,
[... 1641 lines cut ...]
      "Time": "2023-02-21T18:40:10.3921341+07:00",
      "Group": "IPC",
      "Message": "Connected to a client.",
      "StackTrace": null,
      "Level": 0,
      "Notification": false
    },
    {
      "Time": "2023-02-21T18:40:10.3921567+07:00",
      "Group": "Display",
      "Message": "DPI Awareness enabled",
      "StackTrace": null,
      "Level": 0,
      "Notification": false
    },
    {
      "Time": "2023-02-21T18:40:10.3921795+07:00",
      "Group": "Settings",
      "Message": "Presets have been refreshed. Loaded 0 presets.",
      "StackTrace": null,
      "Level": 1,
      "Notification": false
    }
  ]
}

jamesbt365, 2023-02-21:
#2264

closed 2023-02-21 by jamesbt365

Clankx, 2023-02-22:
> https://raw.githubusercontent.com/jamesbt365/OpenTabletDriver/5b7c11fc639fa8cf95a9157ab06326927098d7c1/OpenTabletDriver.Configurations/Configurations/Huion/H640P.json in a folder called Configurations next to otd.

Thank you! It works!