OpenTabletDriver/OpenTabletDriver #2649: Support Request: ArgumentNullException Driver is coming up as null

mailoka113, 2023-02-25:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### What do you need help with?

My tablet is called the Huion New 1060 Plus (Inspiroy) and when I look at supported drivers it says supported. I have tried Huion Drivers to see if the tablet works and it doesnt, so I tried OpenTabletDriver, and this set of errors come up. 

I have attached a couple of images and the JSON file for the diagnostics that I converted to TXT.

[image]

[image]

### What have you tried?

When looking into the issue, I exported the diagnostics and downloaded the configuration for my specific device and found that the Product and vendor ID matched, but everything else on the diagnostics in that little section was named ''Null''

Ive uninstalled and reinstalled drivers, both for huion and OTD. I have restarted many a time. Ive watched MrCubix's video on OTD fix. I downloaded the configuration file for New 1060 Plus.

### Diagnostic Information

```json
{
  "App Version": "OpenTabletDriver v0.6.0.4",
  "Build Date": "5/4/2022",
  "Operating System": {
    "Platform": 2,
    "ServicePack": "",
    "Version": "10.0.19045.0",
    "VersionString": "Microsoft Windows NT 10.0.19045.0"
  },
  "Environment Variables": {
    "USER": null,
    "TEMP": "C:\\Users\\Aaron\\AppData\\Local\\Temp",
    "TMP": "C:\\Users\\Aaron\\AppData\\Local\\Temp",
    "TMPDIR": null,
    "USERPROFILE": "C:\\Users\\Aaron"
  },
  "HID Devices": [
    {
      "DevicePath": "\\\\?\\hid#vid_256c&pid_006e&mi_01#9&2ee3f9f7&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "Manufacturer": null,
      "ProductName": null,
      "SerialNumber": "",
      "FriendlyName": null,
      "VendorID": 9580,
      "ProductID": 110,
      "InputReportLength": 8,
      "OutputReportLength": 0,
      "FeatureReportLength": 0,
      "CanOpen": false
    },
    {
      "DevicePath": "\\\\?\\hid#vid_256c&pid_006e&mi_00#9&6794c2e&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "Manufacturer": null,
      "ProductName": null,
      "SerialNumber": "",
      "FriendlyName": null,
      "VendorID": 9580,
      "ProductID": 110,
      "InputReportLength": 8,
      "OutputReportLength": 0,
      "FeatureReportLength": 0,
      "CanOpen": true
    },
    {
[... 1747 lines cut ...]
      "Message": "Connected to a client.",
      "StackTrace": null,
      "Level": 0,
      "Notification": false
    },
    {
      "Time": "2023-02-25T15:56:12.8392976+00:00",
      "Group": "Display",
      "Message": "DPI Awareness enabled",
      "StackTrace": null,
      "Level": 0,
      "Notification": false
    },
    {
      "Time": "2023-02-25T15:56:12.8393055+00:00",
      "Group": "Settings",
      "Message": "Presets have been refreshed. Loaded 0 presets.",
      "StackTrace": null,
      "Level": 1,
      "Notification": false
    }
  ]
}

```

jamesbt365, 2023-02-25:
tablet > device string reader
9580
110
dump all.

Also the null is normal.

mailoka113, 2023-02-25:
[stringdump.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/10831699/stringdump.txt)

This is what comes up. What do I next?

jamesbt365, 2023-02-25:
https://github.com/OpenTabletDriver/OpenTabletDriver/blob/v0.6.0.4/OpenTabletDriver.Configurations/Configurations/Huion/New%201060%20Plus.json

jamesbt365, 2023-02-25:
https://opentabletdriver.net/Wiki/FAQ/Windows#uclogic-supported-undetected

closed 2023-02-25 by jamesbt365

mailoka113, 2023-02-25:
Ive followed the steps under ''supported tablet not detected'' and my tablet still isnt working/is coming up null
[stillnull.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/10831761/stillnull.txt)

jamesbt365, 2023-02-25:
did you run the application as admin?

mailoka113, 2023-02-25:
Ive tried both, with and without admin and they dont work, when in admin though this comes up on launch that its not meant to be run in admin.

[image]
[image]

jamesbt365, 2023-02-25:
Not OpenTabletDriver. The cleanup. Don't run OpenTabletDriver as admin.

mailoka113, 2023-02-25:
I have, yes. I uninstalled everything.

jamesbt365, 2023-02-25:
Replug your tablet, if it still doesn't work go into device manager > view > device by container, find your tablet > open the drop down menu and screenshot.

mailoka113, 2023-02-25:
Replugged and didnt work, went to device manager, view, then devices by container. 

Previously, Huion HID was here, but since I uninstalled the driver for it, it isnt there. I imagine either Mass storage device (F:) which shows up when the tablet is plugged in is it, but if not, it might be USB composite device. 

[image]

mailoka113, 2023-02-25:
Also Idk how github really works, but I assume closed means that something has been resolved, is it cool if you could open it back up. I really want to figure out how to fix this issue. If thats not what that means then nvm, hopefully it gets sorted soon.

jamesbt365, 2023-02-25:
Join our discord if possible, this will be much faster there.

mailoka113, 2023-02-25:
Can you send over a link? Thanks.

jamesbt365, 2023-02-25:
https://opentabletdriver.net/Discord