OpenTabletDriver/OpenTabletDriver #2652: hid_uclogic is causing tablet to not be detected, FAQ title/description unclear

Neko-Box-Coder, 2023-03-05:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Steps to Reproduce

OS: Arch with Linux kernen 6.0.10

1. Install OpenTabletDriver
2. Open OpenTabletDriver, it shows `No tablets are detected`
3. Manually detects tablets, still shows `No tablets are detected`

### Expected Behavior

Should detect the tablet, if not at least mention `hid_uclogic` could cause **`No tablets are detected` or `ArgumentOutOfRangeException Value range is [0, 15]. (Parameter 'value')`** under the `Device is currently in use` section in FAQ. 

If there's any associated errors with an issue in FAQ, it should be mentioned or quoted for easier search or debug.

### Observed Behavior

Console shows a lot of `ArgumentOutOfRangeException Value range is [0, 15]. (Parameter 'value')`

`No tablets are detected` on a tablet that is listed as supported. 

### Tablet Device

Gaomon PD1560

### Diagnostic Information

[diag.json.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/10892121/diag.json.txt)

jamesbt365, 2023-03-05:
https://opentabletdriver.net/Wiki/FAQ/Linux#argumentoutofrangeexception #2333

closed 2023-03-05 by jamesbt365

Neko-Box-Coder, 2023-03-05:
@jamesbt365 

I understand there are a lot of issues to go through on a daily basis but please read what I have written in the **Expected Behavior**, especially the **second part** of the sentence.

Any of the symptoms I mentioned are nowhere to be found in that section except the URL, hence **`FAQ title/description unclear`** issue title

jamesbt365, 2023-03-05:
Should detect the tablet **it literally can't, it doesn't have permission to because the kernel driver has it.**
And mentioning it falls under the other issue i pointed to, and things for the wiki should be put on the correct repo, not this one.

rytrix, 2024-06-20:
For anyone having this problem you can fix it by removing the hid_uclogic kernel driver with modprobe.