OpenTabletDriver/OpenTabletDriver #2655: Add support for M5/SN540

mailset, 2023-03-10:
### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Manufacturer

Gaomon

### Product Name

M5/SN540

### Device Width

187mm

### Device Height

127mm

### Pressure Levels

8192

### Lines per Inch (LPI)

5080

### Additional Information

M5 is SN540, but it has two different names

### Diagnostic Information

{
  "App Version": "OpenTabletDriver v0.6.0.4",
  "Build Date": "11/15/2022",
  "Operating System": {
    "Platform": 4,
    "ServicePack": "",
    "Version": "6.2.2.2",
    "VersionString": "Unix 6.2.2.2"
  },
  "Environment Variables": {
    "USER": "mailset",
    "TEMP": null,
    "TMP": null,
    "TMPDIR": null,
    "DISPLAY": ":1",
    "WAYLAND_DISPLAY": "wayland-1",
    "PWD": "/usr/share/OpenTabletDriver",
    "PATH": "/home/mailset/.cargo/bin:/usr/local/sbin:/usr/local/bin:/usr/bin:/home/mailset/.dotnet/tools:/usr/lib/jvm/default/bin:/usr/bin/site_perl:/usr/bin/vendor_perl:/usr/bin/core_perl:/var/lib/snapd/snap/bin:/home/flicko/.local/bin"
  },
  "HID Devices": [
    {
      "DevicePath": "/sys/devices/pci0000:00/0000:00:01.2/0000:12:00.0/usb3/3-6/3-6:1.0/0003:1038:0615.0002/hidraw/hidraw1",
      "Manufacturer": "SteelSeries",
      "ProductName": "SteelSeries Apex M650",
      "SerialNumber": "00000000001A",
      "FriendlyName": "SteelSeries Apex M650",
      "VendorID": 4152,
      "ProductID": 1557,
      "InputReportLength": -1,
      "OutputReportLength": -1,
      "FeatureReportLength": -1,
      "CanOpen": true
    },
    {
      "DevicePath": "/sys/devices/pci0000:00/0000:00:01.2/0000:12:00.0/usb3/3-6/3-6:1.1/0003:1038:0615.0003/hidraw/hidraw2",
[... 1794 lines cut ...]
      "Time": "2023-03-10T21:31:04.9882848+08:00",
      "Group": "IPC",
      "Message": "Disconnected from a client.",
      "StackTrace": null,
      "Level": 0,
      "Notification": false
    },
    {
      "Time": "2023-03-10T21:31:04.9882948+08:00",
      "Group": "IPC",
      "Message": "Connected to a client.",
      "StackTrace": null,
      "Level": 0,
      "Notification": false
    },
    {
      "Time": "2023-03-10T21:31:04.9883061+08:00",
      "Group": "Settings",
      "Message": "Presets have been refreshed. Loaded 0 presets.",
      "StackTrace": null,
      "Level": 1,
      "Notification": false
    }
  ]
}

jamesbt365, 2023-04-07:
tablet > device string reader
9580
100
dump all, save that file, send it here.

telecomadm1145, 2023-04-23:
201: OEM02_T195_210513
[M5.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/11302864/M5.txt)

jamesbt365, 2023-08-23:
Incomplete pull request at #2831 should fix this.

PR #3056 (2025-09-21): Add Gaomon M5 support
Freshens up stale PR at #2831

Fixes #2655

No verification, so setting as draft until tested
closed 2025-09-21 by Kuuuube