OpenTabletDriver/OpenTabletDriver #2664: [Plugin:Error]	Unable to construct object 'OpenTabletDriver.Desktop.Output.AbsoluteMode'

yue-dongchen, 2023-03-17:
I'm on Wayland Kubuntu 22.10. The relative mode works, but the above error is given when I switch to absolute or artist mode. The tablet is completely unresponsive.

jamesbt365, 2023-03-17:
As you decided to ignore the templates, help > export diagnostic, save the file and send it here.

yue-dongchen, 2023-03-18:
Yeah, my bad. I missed it. Later I ungraded Plasma to 5.26, which has the Wayland drawing tablet kcm. I'll close this as no one else seems to have this problem.

sbstratos79, 2023-03-22:
Could you please re-open the issue. I am also getting this error. Here's the log:
[otd.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/11041253/otd.txt)

X9VoiD, 2023-03-22:
@sbstratos79 you don't have the same underlying issue. Make sure that the user you're running the daemon from has access to `/run/user/1000/:0`.

closed 2023-03-22 by X9VoiD