OpenTabletDriver/OpenTabletDriver #2670: No input after waking from sleep (linux)

Agreeable-f, 2023-03-20:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Steps to Reproduce

1. turn on pc and log in. (not sure if it is required to plug in the tablet first for this to happen) 
2. put pc to sleep (for me i clicked log out > sleep)
3. turn pc back on and log in

### Expected Behavior

cursor movement from my tablet after waking my pc from sleep

### Observed Behavior

my tablet (ctl 472) detects input shown on the led but my cursor isn't moved on my pc

### Tablet Device

Wacom CTL-472

### Diagnostic Information

[OTD_diagnostics.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/11021439/OTD_diagnostics.txt)

Agreeable-f, 2023-03-20:
i'd like to say yes. most recent aur package
[image]

jamesbt365, 2023-03-20:
Thats not master, that's the latest release. Master is latest git.

Agreeable-f, 2023-03-23:
doesn't seem to happen on latest master as of this comment but latest master also did this when loading my area
[image]

Agreeable-f, 2023-03-23:
it's also unable to properly load stable's settings.json. just sets everything to either nan or 0

X9VoiD, 2023-03-23:
> ... but latest master also did this when loading my area

Known issue: #2409

> it's also unable to properly load stable's settings.json. just sets everything to either nan or 0

That's expected for now, master changed settings format.

X9VoiD, 2023-03-23:
Will close since this is not reproducible on `master`

closed 2023-03-23 by X9VoiD