OpenTabletDriver/OpenTabletDriver #2672: Support Request:

Koizaydee, 2023-03-22:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### What do you need help with?

tablet cursor stops and teleports to another position based on the pen's current position

### What have you tried?

Reinstalling the drivers
Deleting the data folder in Appdata\Local
Changing the cable
Turning off Raw Input in osu
Switched from stable to cutting-edge and switched back
Turning off all filters

### Diagnostic Information

[diagnostic info.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/11037896/diagnostic.info.txt)

jamesbt365, 2023-03-22:
Could you get a video? From what you have described that seems normal. Also I don't know about you but that certainly isn't a "concise and meaningful title.". Please do not randomly click boxes.

Koizaydee, 2023-03-23:
ive tested it but now it was working fine for now, so ill close this issue