OpenTabletDriver/OpenTabletDriver #2673: Support Request: DTK-1661（1660 Chinese version）

song2daodao, 2023-03-22:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### What do you need help with?

DTK-1661 is a version of 1660 but sold in China, exactly same as 1660, only the model number is different, please consider to support, thanks.

### What have you tried?

The driver is installed, but the DTK-1661 tablet is not detected.

### Diagnostic Information

-

jamesbt365, 2023-03-22:
Wrong template. Also can't support with a diagnostic, which you decided to not supply.

song2daodao, 2023-03-22:
> Wrong template. Also can't support with a diagnostic, which you decided to not supply.

**diagnostic**：
[0000.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/11040743/0000.txt)

I'm so sorry, because the tablet is not detected, so I thought providing diagnostic wouldn't help. And, I found that dtk-1661 is detected as dtk-1660 in the new version of wacom driver, so maybe my question should be replaced with: Why my cintiq16 is not detected by the driver.

jamesbt365, 2023-04-07:
The tablet was supported in #2384 and will be present in the next release of opentabletdriver, alternatively, you can download the file from [here](https://raw.githubusercontent.com/OpenTabletDriver/OpenTabletDriver/c8581cce76d491226645e27e6cb6a356e3a0c04b/OpenTabletDriver.Configurations/Configurations/Wacom/DTK-1660.json) and put this file into a folder called `Configurations` next to otd.

closed 2023-04-07 by jamesbt365