OpenTabletDriver/OpenTabletDriver #2685: Display Toggle

websterek, 2023-03-30:
### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Description

Greetings!

Please add keyboard hot-key to toggle current tablet display. On dual (or multi) monitor setup it's very convenient to switch active screen with one hand laying on keyboard and second holding pen on the tablet.

For now there are few "hacks" using AutoHotkey (for Wacom) such as: https://kimstrandli.com/kswacomdisplaytoggle/ and they works pretty good. Still it's will be awesome to have such option natively.

Best!

jamesbt365, 2023-03-30:
This exists in master through bindings to switch preset

websterek, 2023-03-30:
> This exists in master through bindings to switch preset

Oh, I couldn't find it, it's somehow system related?
I'm using Windows btw.

jamesbt365, 2023-03-30:
file > save as preset then you can set a binding to switch the preset. That exists currently only on master, which doesn't have any plugins and is a little broken otherwise, I don't recommend using master currently but the option exists.

websterek, 2023-03-30:
> 
Sure thing, but for tests I have downloaded master. I can add display toggle directly to tablet express keys (auxiliary) but there is no way to add toggle to keyboard shortcut?

jamesbt365, 2023-08-23:
I'm going to close as this feature was added in the DI rewrite (preset switching on binding), as well as a plugin existing for support in 0.6.1+ called Monitor Toggle, which can be located in the plugin manager.

closed 2023-08-23 by jamesbt365

websterek, 2024-08-15:
I known issu is closed for now, but we have some misunderstand (probably). I don't mean to assign "screen toggle" to tablet buttons but user keyboard for example using key combination [Super] + [Space]. For now I'm using bash script assigned to shortcut trought KDE as workaround:

```
#!/bin/bash

# File to store the last executed command
TOGGLE_FILE=~/.wacom_toggle_state

# Command 1
CMD1='otd setdisplayarea "Wacom CTL-6100WL" 2560 1440 1280 720'
# Command 2
CMD2='otd setdisplayarea "Wacom CTL-6100WL" 2560 1440 3840 702'

# Check if the toggle file exists, and create it if it doesn't
if [ ! -f $TOGGLE_FILE ]; then
    echo "1" > $TOGGLE_FILE
fi

# Read the current state from the file
STATE=$(cat $TOGGLE_FILE)

# Toggle between the two commands
if [ "$STATE" = "1" ]; then
    eval $CMD2
    echo "2" > $TOGGLE_FILE
else
    eval $CMD1
    echo "1" > $TOGGLE_FILE
fi
```

And it is working correctly, besides the delay caused by the time needed for the otd to be executed. This would be a very appreciated feature for graphic designers who have multiple screens. A similar solution was natively implemented by the official Huion drivers a few months ago: [https://support.huion.com/en/support/solutions/articles/44002399072-how-to-use-hotkeys-to-switch-screens](https://support.huion.com/en/support/solutions/articles/44002399072-how-to-use-hotkeys-to-switch-screens)

jamesbt365, 2024-08-15:
Global WM keybinds are on the backlog and will appear sometime after the release of 0.7.