OpenTabletDriver/OpenTabletDriver #2693: tablet not detected wacom ctl 472

AndreasPlysis, 2023-04-02:
[Detect:Warning]	Another tablet driver found: Wacom
[Detect:Info]	Searching for tablets...
[Detect:Info]	No tablets were detected.
[Settings:Info]	Presets have been refreshed. Loaded 0 presets.
[Detect:Info]	Searching for tablets...
[Detect:Info]	No tablets were detected.
[Detect:Info]	Searching for tablets...
[Detect:Info]	No tablets were detected.

it was half working before with the driver it was able to detect it but after i uninstalled the driver it cant even detect it. does anyone know the fix for this one?

jamesbt365, 2023-04-02:
https://github.com/X9VoiD/TabletDriverCleanup/releases/tag/v4.0.1

jamesbt365, 2023-04-02:
If the issue persists provide a diagnostic with help > export diagnostic, save the file and send it here. Next time also use the support request template, it is much better for such a situation.

closed 2023-04-02 by jamesbt365