OpenTabletDriver/OpenTabletDriver #2695: OTD uses spyware

navid-zamani, 2023-04-07:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Steps to Reproduce

Just follow the official installation instructions for Ubuntu.

### Expected Behavior

Get a normal Linux kernel driver and maybe a user mode configuration facility (say a file/directory in /etc and an init script or configuration tool).

### Observed Behavior

Notice the instructions say `sudo apt install -y dotnet-sdk-6.0`.
Notice that when running that, you also get a info that says

> Telemetry
> ---------
> The .NET tools collect usage data in order to help us improve your experience. It is collected by Microsoft and shared with the community. You can opt-out of telemetry by setting the DOTNET_CLI_TELEMETRY_OPTOUT environment variable to '1' or 'true' using your favorite shell.
> 
> Read more about .NET CLI Tools telemetry: https://aka.ms/dotnet-cli-telemetry

What the actual fuck??

Why do you think everyone is installing Linux right now?  
To get away from *exactly* this sort of criminal behavior!  
And yes, in the EU, is is *actually* a crime. [See: GDPR.] Especially with your instructions including a `-y` to opt in without even asking the victim.

You claim to be an open driver for Linux. Yet, not even one of those three things is correct! You are not open (.NET), it is not a driver (it is basically the modern equivalent of a DOS TSR, not a file in /lib/modules/*/ or /usr/src/*linux*/ for example), and it is very very much not Linux (.NET).

It is literally spyware and literally a crime.

This is seriously *not cool*! What were you thinking?? You clearly are new to this Linux thing, and like other groups coming from Windows, think you can just take these Windows ways and force them onto our community?

I recommend fixing this or retracting your project, as otherwise I will have to report you to law enforcement. I’m sorry, but if all of this was just an  April fools joke, this is completely not funny.

### Tablet Device

Huion WL1409 V2

### Diagnostic Information

Impossible to provide without installing MALWARE.

32-bit-link, 2023-04-07:
> for Ubuntu.

Doesn't ubuntu have telemetry? 

Please do report this (and many other """"open source"""" projects) to law enforcement as spying is incredibly illegal and I cannot believe that a so called "open" driver would do such a thing!

adryzz, 2023-04-07:
>  You are not open (.NET)

https://github.com/dotnet/runtime

> it is not a driver
[image]
https://en.wikipedia.org/wiki/Device_driver

> very very much not Linux (.NET).
https://github.com/dotnet/core/blob/main/release-notes/7.0/supported-os.md#linux

also you use ubuntu, you're one to talk about telemetry

closed 2023-04-07 by X9VoiD