OpenTabletDriver/OpenTabletDriver #2696: Support Request: I have the same issue as #1396 but didn't sloved

akisa18, 2023-04-09:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### What do you need help with?

the issue start with some strange teleporting with the cursor, which is it works fine for about 10mins then it start to stick at the top(I'm sure that I don't have any other drivers running). I thought it is my tablet issue so I planned to switch another tablet but things get worst.
[image]
[image]

these are some of the problems that pops up, I apologize to not have these completely. I tried reinstalling, didn't work and then I go on the wiki page and tried the winusb download. Now it comes to the #1396 problem.

### What have you tried?

- restarting
- reinstalling
- download the wacom offical driver(worked strangely with wacom mouse mode with otd opened)
- go to the control panel to reinstall all of the drivers
- winusb download
- all of the existing not detecting issues solutions. 

### Diagnostic Information

[1.json.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/11184560/1.json.txt)

jamesbt365, 2023-04-09:
> didn't work and then I go on the wiki page and tried the winusb download.

There is your issue, you installed winusb, as the wiki page says do not install winusb unless the tablets need it, it then provides you with a list of tablets that need it.

Go into device manager > view > device by container > find your tablet > open the container > find the CTL-472's inside > right click them > update > browse my computer > let me pick from a list of available drivers > usb input/composite device > next

if you cannot find this, screenshot the device container with it open and I will assist you further from here.

akisa18, 2023-04-09:
it works! can I leave this open just in case it start to cursor teleport again?

jamesbt365, 2023-04-09:
The only reason it would teleport is if you have invalid settings or have another driver installed, you can check with tabletdrivercleanup, which is on the wiki. You may also find that it is easier to ask for help over our [Discord](https://opentabletdriver.net/discord).

closed 2023-04-09 by jamesbt365