OpenTabletDriver/OpenTabletDriver #2697: Add: support for tablet buttons for wacom PTZ-431W

SmoczyWasTaken, 2023-04-09:
### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Description

I would like support for touchbar/tablet buttons for Wacom PTZ-431W

jamesbt365, 2023-08-23:
This should have been fixed ages back, I don't see a reason why it wouldn't be because the rest of the intuos3 tablets now have working aux. I can't find the pull request at the moment but please reopen if this doesn't work on the latest release.

jamesbt365, 2023-08-23:
Oh, and touchbars won't work entirely because OTD doesn't support this. This is listed on our tablets.md and we are aware about the missing support (otd doesn't support touch bars at all, not tablet specific)

closed 2023-08-23 by jamesbt365