OpenTabletDriver/OpenTabletDriver #2700: Support Request: HUION RTE 100

1Oregu1, 2023-04-13:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### What do you need help with?

bought a new tablet, but did not check if it is in the list of supported tablets
HUION RTE 100, its not supported by OTB

### What have you tried?

tried nothing, cuz what can i do, if its not supported

### Diagnostic Information

sorry, i don`t know what put in here

jamesbt365, 2023-04-13:
Diagnostic information is required for supporting tablets. As you missed the instructions to do so go to help > export diagnostics, save that file and upload it here. Once you have done that, we can get started.

1Oregu1, 2023-04-13:
oh, thanks, here u go
{
  "App Version": "OpenTabletDriver v0.6.0.4",
  "Build Date": "5/4/2022",
  "Operating System": {
    "Platform": 2,
    "ServicePack": "",
    "Version": "10.0.22000.0",
    "VersionString": "Microsoft Windows NT 10.0.22000.0"
  },
  "Environment Variables": {
    "USER": null,
    "TEMP": "C:\\Users\\USER\\AppData\\Local\\Temp",
    "TMP": "C:\\Users\\USER\\AppData\\Local\\Temp",
    "TMPDIR": null,
    "USERPROFILE": "C:\\Users\\USER"
  },
  "HID Devices": [
    {
      "DevicePath": "\\\\?\\hid#vid_258a&pid_1006&mi_01&col08#8&2602e0ce&0&0007#{4d1e55b2-f16f-11cf-88cb-001111000030}\\kbd",
      "Manufacturer": "Gaming KB ",
      "ProductName": "Gaming KB ",
      "SerialNumber": "",
      "FriendlyName": "Gaming KB ",
      "VendorID": 9610,
      "ProductID": 4102,
      "InputReportLength": 8,
      "OutputReportLength": 0,
      "FeatureReportLength": 0,
      "CanOpen": false
    },
    {
      "DevicePath": "\\\\?\\hid#vid_04d9&pid_fc5f&mi_02&col02#8&32b84a72&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "Manufacturer": "Unknown Manufacturer",
      "ProductName": "USB Gaming Mouse",
      "SerialNumber": "",
      "FriendlyName": "USB Gaming Mouse",
      "VendorID": 1241,
      "ProductID": 64607,
      "InputReportLength": 8,
      "OutputReportLength": 0,
      "FeatureReportLength": 2048,
      "CanOpen": true
    },
    {
      "DevicePath": "\\\\?\\hid#vid_258a&pid_1006&mi_01&col02#8&2602e0ce&0&0001#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "Manufacturer": "Gaming KB ",
      "ProductName": "Gaming KB ",
      "SerialNumber": "",
      "FriendlyName": "Gaming KB ",
      "VendorID": 9610,
      "ProductID": 4102,
      "InputReportLength": 3,
      "OutputReportLength": 0,
      "FeatureReportLength": 0,
      "CanOpen": true
    },
    {
      "DevicePath": "\\\\?\\hid#vid_04d9&pid_fc5f&mi_01#8&ee10cb0&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}\\kbd",
      "Manufacturer": "Unknown Manufacturer",
      "ProductName": "USB Gaming Mouse",
      "SerialNumber": "",
      "FriendlyName": "USB Gaming Mouse",
      "VendorID": 1241,
      "ProductID": 64607,
      "InputReportLength": 9,
      "OutputReportLength": 0,
      "FeatureReportLength": 0,
      "CanOpen": false
    },
[... 1784 lines cut ...]
      "Time": "2023-04-13T16:07:27.178854+03:00",
      "Group": "IPC",
      "Message": "Connected to a client.",
      "StackTrace": null,
      "Level": 0,
      "Notification": false
    },
    {
      "Time": "2023-04-13T16:07:27.1788617+03:00",
      "Group": "Display",
      "Message": "DPI Awareness enabled",
      "StackTrace": null,
      "Level": 0,
      "Notification": false
    },
    {
      "Time": "2023-04-13T16:07:27.1788694+03:00",
      "Group": "Settings",
      "Message": "Presets have been refreshed. Loaded 0 presets.",
      "StackTrace": null,
      "Level": 1,
      "Notification": false
    }
  ]
}

jamesbt365, 2023-04-13:
Could you please edit your message so that it contains the file instead of dumping it here, it makes it hard to read. you can change the file extension to txt and upload it by just dragging it in.

jamesbt365, 2023-04-13:
Alternatively, we have a [Discord](https://opentabletdriver.net/discord) server and this would speed up creation if you are available on that platform.

1Oregu1, 2023-04-13:
ok, i`ll gonna send it in discord

PR #2701 (2023-08-13): Add support for Huion RTE-100
Fixes #2700 

Verification: https://discord.com/channels/615607687467761684/615611007951306863/1096096274375250050
Diagnostic: [diag4.json](https://github.com/OpenTabletDriver/OpenTabletDriver/files/11223960/diag4.json.txt)
closed 2023-08-13 by X9VoiD