OpenTabletDriver/OpenTabletDriver #2708: Provide flatpak of OpenTabletDriver

JamiKettunen, 2023-04-20:
### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Description

Providing an official `net.opentabletdriver.OpenTabletDriver` flatpak on https://flathub.org would be pretty nice for a cross-distribution solution; the prebuilt binaries provided with releases only run on glibc distributions and on some distributions .NET SDK/runtime may not be easily available (the bootstrapping process isn't exactly fun either).

Even on glibc [Void Linux](https://voidlinux.org) I've manually extracted the OTD and .NET runtime binaries and set up the necessary integrations to make them work nicely without being managed by anything.

One issue would be the udev rules similar to the manual tarball setup, but a message about setting them up on launch similar to [the Steam flatpak](https://github.com/flathub/com.valvesoftware.Steam/blob/7b0c8dc/steam_wrapper/steam_wrapper.py#L141-L149) could be added for example.

https://github.com/flathub/com.github.PintaProject.Pinta could be used as an example for a flatpak including the .NET runtime along with some program.

As for auto-starting, keeping in line with the distro (and init system) agnostic nature of flatpak, an [XDG autostart](https://specifications.freedesktop.org/autostart-spec/autostart-spec-latest.html) `.desktop` file could be created (on first launch?) which starts the `OpenTabletDriver.Daemon`.

Thoughts? I'm considering taking a shot at attempting this myself sooner or later :)

X9VoiD, 2023-04-20:
Ah seems like I missed the part where you addressed that.

InfinityGhost, 2024-03-27:
Will be closed by https://github.com/flathub/flathub/pull/5082

jamesbt365, 2024-05-14:
https://github.com/flathub/net.opentabletdriver.OpenTabletDriver/

closed 2024-05-14 by jamesbt365