OpenTabletDriver/OpenTabletDriver #2715: Nuget release schedule

peppy, 2023-04-24:
Any chance of having a more frequent nuget release schedule? We get occasional cases of issues which have been fixed years ago but not included in a release (recent case being https://github.com/ppy/osu/issues/23289), would be very nice to have things pushed automatically (or semi-automatically) on a more regular schedule.

peppy, 2023-06-12:
Any updates here? Would it be easier if we start to build our own nuget package if we want more frequent releases?

X9VoiD, 2023-06-12:
Currently dealing with meta issues like packaging to eventually speed up releases, `hotfix` branch is where the effort is going for now (`master` is going to be in a bad state for a while so we forked off `0.6.0.4` and backported important changes) and for the most part it is "ready" with the only remaining issues being a meta one or something that only affects full fat OTD. We might be able to do a 0.6.1.0 nuget release out of hotfix branch this week so may probably hold off on building downstream nuget for now.

jamesbt365, 2025-01-24:
We have CI to publish nuget automatically on release (#3555 and #3550), (and we are doing more frequent, small releases now) so this should be solved. Please reopen if I'm missing anything.

closed 2025-01-24 by jamesbt365