OpenTabletDriver/OpenTabletDriver #2716: [Veikk A50] Cursor teleports while using relative mode.

AsteroidzX, 2023-04-24:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Steps to Reproduce

https://user-images.githubusercontent.com/30378525/234105792-65f06f84-add3-476e-a557-5ae4c7140791.mp4

brining the pen from outside the bounds of the active location seems to be the best way to demonstrate the bug, from whichever side i bring the pen inside the bounds it teleports the cursors location to where it's placed instead of continuing its motion from the cursors previous location.

### Expected Behavior

continuation of motion from previous position of cursor

### Observed Behavior

teleportation of cursor to a different location.

### Tablet Device

Veikk A50

### Diagnostic Information

[diag.txt.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/11314493/diag.txt.txt)

jamesbt365, 2023-04-24:
If you don't want this behavior you have to lower the reset time.

closed 2023-04-24 by jamesbt365

AsteroidzX, 2023-04-24:
> If you don't want this behavior you have to lower the reset time.

thanks a ton my dude, switched it to 5ms and problem got solved. thanks again man <3