OpenTabletDriver/OpenTabletDriver #2717: Support Request Tablet not detected Huion 950p

Nheptune, 2023-04-25:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### What do you need help with?

Tablet not being detected

### What have you tried?

1.Ran TabletDriverCleaner
2.Exported Diagnostics:
3.Changed the config identifier
4.Configs show up but still not being detected
5.device string reader doesn't create dump.txt

### Diagnostic Information

[Diagnostics1.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/11319865/Diagnostics1.txt)

jamesbt365, 2023-08-23:
tablet > device string reader
9580
100

dump all (leave the bottom box blank, its not important)

save that file and send it here.

jamesbt365, 2023-12-26:
@Nheptune just a reminder that we need this information in order to fix this issue.

closed 2024-07-30 by jamesbt365