OpenTabletDriver/OpenTabletDriver #2721: Support Request: OpenTabletDriver (Linux) cant find my Gaomon s620 Tablet

AshnOnee, 2023-04-26:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### What do you need help with?

OpenTabletDriver cant find my Gaomon s620 Tablet, Linux O.C- Nobara 37/Fedora

### What have you tried?

Couldnt find any help on youtube etc.

### Diagnostic Information

Cant do this idk

jamesbt365, 2023-04-26:
The diagnostic information page is not optional for diagnosing tablet issues. Go to help > export diagnostic, save that file and send it here.

jamesbt365, 2023-08-23:
This seems to have gone stale so either this was because the s620 new variant (which has a configuration in the latest releases) or a udev/hid_uclogic kernel module interference, both of which have solutions listed on our [FAQ](https://opentabletdriver.net/Wiki/FAQ/Linux).

closed 2023-08-23 by jamesbt365