OpenTabletDriver/OpenTabletDriver #2726: Huion Q620M dial support

J4ckTh3R1pper, 2023-05-04:
### Acknowledgements

- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Description

I am using Huion Q620M, all physical buttons worked flawlessly out of the box, however the Dial ( a scroll wheel ) isn't working, I tried debugging tablet in the GUI and found that the rolling events are actually recognized as AuxButtons by the driver, but in the 'Auxiliary Settings', there are only 8 sections that are related to 8 physical buttons on the tablet. Is there a way to make use of the Dial ?

Raw Tablet Data
when only scrolling clockwise: 
08 F1 01 01 00 01 00 00 00 00 00 00
when only scrolling counterclockwise: 
08 F1 01 01 00 02 00 00 00 00 00 00
after pressing down dial and before next dial input
08 E0 01 01 00 00 00 0088 00 E7 00

Tablet Report
When scrolling clockwise:
[False False False False False False False False True False False False False]
When scrolling counterclockwise:
[False False False False False False False False False True False False False]
When Pressing:

jamesbt365, 2023-05-04:
wheel support is tracked with #1367 already and a partial implementation of wheels is in #2669 closing as duplicate.

closed 2023-05-04 by jamesbt365

gonX, 2026-03-04:
As for the wheel button, this needs to be handled separately. I'll need a full tablet data recording of you pressing all aux buttons in order. Avoid using your pen while recording, to keep the recording concise.

You can record the raw data by opening the OpenTabletDriver interface, and then going to Tablets -> Tablet Debugger. There will then be a "Enable Data Recording" checkbox on the right hand of the new window. Then a `tablet-data`  file will be available in your [app data](https://opentabletdriver.net/Wiki/FAQ/General#appdata) folder.

NeoTheFox, 2026-03-06:
[tablet-data_1772803353.txt](https://github.com/user-attachments/files/25795877/tablet-data_1772803353.txt)
@gonX Here's the tablet-data log of me only spinning the dial and pressing the button at the end.