OpenTabletDriver/OpenTabletDriver #2728: PrintScreen not working as an auxiliary button

aL1NaZ4R, 2023-05-04:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Steps to Reproduce

1. Install Lightshot and define "PrintScreen" as hotkey.
2. Press PrintScreen button on the keyboard and watch it's working.
3. Now go to Auxiliary buttons in OTD
4. Define a button for "PrintScreen"
[image]
5. Then when you press that button, nothing will happen.

### Expected Behavior

To work and Screenshot tool come up.

### Observed Behavior

Nothing happens.

### Tablet Device

XP-Pen G640S (Not V2)

### Diagnostic Information

[696969696.txt](https://github.com/OpenTabletDriver/OpenTabletDriver/files/11400159/696969696.txt)

X9VoiD, 2023-05-22:
Resolved by #2739 and #2740

closed 2023-05-22 by X9VoiD

aL1NaZ4R, 2023-05-23:
> Resolved by #2739 and #2740

Cheers!