OpenTabletDriver/OpenTabletDriver #2732: my tablet (huion 420) not detected anymore

Gialemba, 2023-05-08:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Steps to Reproduce

1. everything worked fine yesterday
2. launch the pc and launch the cmd driver by accident
3. close it (first by trying to type exit but nothing happen so just click on the X) cause i wanted to open the gui version
4. launch the gui version
5. still detected but got issue with the usability
6. restart the pc
7. launch the gui version only
8. my tablet is not detected anymore
9. go to the console section to see some log and a warning said i got huion driver detected and it may not work cause of it
10. did not found a huion intaller/uninstaller to remove it "clean" on my pc ('cause i already did it when i installed this one)
11. go to the control panel and delete the only driver i found named huion driver and then delete it
12. restart my pc and try again
13. same console message
14. come here and ask for the issue solution

### Expected Behavior

i expect it work like yesterday (fine)

### Observed Behavior

the issue i got first was like my pen got detected on the tablette only a few time like it was at 2 fps and only on few spot not all the aera i selected on the gui but now it just not detected anymore so i can't use it

### Tablet Device

huion 420

### Diagnostic Information

```json
{
  "App Version": "OpenTabletDriver v0.6.0.4",
  "Build Date": "5/4/2022",
  "Operating System": {
    "Platform": 2,
    "ServicePack": "",
    "Version": "10.0.22621.0",
    "VersionString": "Microsoft Windows NT 10.0.22621.0"
  },
  "Environment Variables": {
    "USER": null,
    "TEMP": "C:\\Users\\cbvan\\AppData\\Local\\Temp",
    "TMP": "C:\\Users\\cbvan\\AppData\\Local\\Temp",
    "TMPDIR": null,
    "USERPROFILE": "C:\\Users\\cbvan"
  },
  "HID Devices": [
    {
      "DevicePath": "\\\\?\\hid#elan1200&col01#4&308e827&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "Manufacturer": "Microsoft",
      "ProductName": "HIDI2C Device",
      "SerialNumber": "9999",
      "FriendlyName": "HIDI2C Device",
      "VendorID": 1267,
      "ProductID": 12474,
      "InputReportLength": 9,
      "OutputReportLength": 0,
      "FeatureReportLength": 0,
      "CanOpen": false
    },
    {
[... 1813 lines cut ...]
      "Group": "Settings",
      "Message": "Presets have been refreshed. Loaded 0 presets.",
      "StackTrace": null,
      "Level": 1,
      "Notification": false
    },
    {
      "Time": "2023-05-08T11:53:05.73854+01:00",
      "Group": "Tablet",
      "Message": "Tablet debugging is enabled",
      "StackTrace": null,
      "Level": 0,
      "Notification": false
    },
    {
      "Time": "2023-05-08T11:53:05.7385525+01:00",
      "Group": "Tablet",
      "Message": "Tablet debugging is disabled",
      "StackTrace": null,
      "Level": 0,
      "Notification": false
    }
  ]
}
```

jamesbt365, 2023-05-08:
You have the huion driver installed and I don't see your tablet plugged in.

Gialemba, 2023-05-08:
my tablet is for sure plugged in 'cause it got the green led when i plug it in) and the huion driver (installed or not) was there yesterday too but it was working fine

jamesbt365, 2023-05-08:
https://github.com/X9VoiD/TabletDriverCleanup/releases/latest.

Gialemba, 2023-05-08:
i just did it and then restart it but it's still not detected (may it be my tablette who just died ?) here the new diagnostic files : 
```json
{
  "App Version": "OpenTabletDriver v0.6.0.4",
  "Build Date": "5/4/2022",
  "Operating System": {
    "Platform": 2,
    "ServicePack": "",
    "Version": "10.0.22621.0",
    "VersionString": "Microsoft Windows NT 10.0.22621.0"
  },
  "Environment Variables": {
    "USER": null,
    "TEMP": "C:\\Users\\cbvan\\AppData\\Local\\Temp",
    "TMP": "C:\\Users\\cbvan\\AppData\\Local\\Temp",
    "TMPDIR": null,
    "USERPROFILE": "C:\\Users\\cbvan"
  },
  "HID Devices": [
    {
      "DevicePath": "\\\\?\\hid#hidclass#1&2d595ca7&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "Manufacturer": "Shaul Eizikovich",
      "ProductName": "vJoy - Virtual Joystick",
      "SerialNumber": "2.1.9",
      "FriendlyName": "vJoy - Virtual Joystick",
      "VendorID": 4660,
      "ProductID": 48813,
      "InputReportLength": 65,
      "OutputReportLength": 18,
      "FeatureReportLength": 5,
      "CanOpen": true
    },
    {
      "DevicePath": "\\\\?\\hid#vid_04d9&pid_0295&mi_01#8&1b2d87d9&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "Manufacturer": "Unknown Manufacturer",
      "ProductName": "CYRM02p0295h00E0r0101",
      "SerialNumber": "",
      "FriendlyName": "CYRM02p0295h00E0r0101",
      "VendorID": 1241,
      "ProductID": 661,
      "InputReportLength": 65,
      "OutputReportLength": 65,
      "FeatureReportLength": 0,
      "CanOpen": true
    },
    {
      "DevicePath": "\\\\?\\hid#vid_256c&pid_006e&mi_01&col01#9&31e5df41&0&0000#{4d1e55b2-f16f-11cf-88cb-001111000030}",
      "Manufacturer": "HUION",
      "ProductName": "420",
      "SerialNumber": "",
      "FriendlyName": "420",
      "VendorID": 9580,
      "ProductID": 110,
      "InputReportLength": 8,
      "OutputReportLength": 0,
      "FeatureReportLength": 2,
      "CanOpen": false
    },
    {
      "DevicePath": "\\\\?\\hid#vid_04d9&pid_0295&mi_02&col04#8&8a9b5e9&0&0003#{4d1e55b2-f16f-11cf-88cb-001111000030}\\kbd",
      "Manufacturer": "Unknown Manufacturer",
      "ProductName": "USB-HID Keyboard",
      "SerialNumber": "",
      "FriendlyName": "USB-HID Keyboard",
      "VendorID": 1241,
      "ProductID": 661,
      "InputReportLength": 31,
      "OutputReportLength": 0,
      "FeatureReportLength": 0,
      "CanOpen": false
[... 1848 lines cut ...]
      "Group": "IPC",
      "Message": "Connected to a client.",
      "StackTrace": null,
      "Level": 0,
      "Notification": false
    },
    {
      "Time": "2023-05-08T12:20:53.7931236+01:00",
      "Group": "Display",
      "Message": "DPI Awareness enabled",
      "StackTrace": null,
      "Level": 0,
      "Notification": false
    },
    {
      "Time": "2023-05-08T12:20:53.7931338+01:00",
      "Group": "Settings",
      "Message": "Presets have been refreshed. Loaded 0 presets.",
      "StackTrace": null,
      "Level": 1,
      "Notification": false
    }
  ]
}
```

jamesbt365, 2023-05-08:
you need to install winusb on interface 0 again.

closed 2023-05-08 by jamesbt365