OpenTabletDriver/OpenTabletDriver #2733: OpenTabletDriver Doesn't start after installation

Shizara, 2023-05-08:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Steps to Reproduce

[image]

### Expected Behavior

It should have opened

### Observed Behavior

It doesn't want to open

### Tablet Device

Wacom CTH-680

### Diagnostic Information

N/A

jamesbt365, 2023-05-08:
Logs from the daemon please.

Shizara, 2023-05-08:
5/8/2023 6:43:29 PM
System.DllNotFoundException
Unable to load shared library 'libX11' or one of its dependencies. In order to help diagnose loading problems, consider setting the LD_DEBUG environment variable: liblibX11: cannot open shared object file: No such file or directory
OpenTabletDriver.Native
   at OpenTabletDriver.Native.Linux.Xorg.XLib.sys_XOpenDisplay(Char* display)
   at OpenTabletDriver.Native.Linux.Xorg.XLib.XOpenDisplay(Char* display)
   at OpenTabletDriver.Desktop.Interop.Display.XScreen..ctor()
   at OpenTabletDriver.Desktop.Interop.DesktopInterop.ConstructLinuxDisplay()
   at OpenTabletDriver.Desktop.Interop.DesktopInterop.get_VirtualScreen()
   at OpenTabletDriver.Desktop.Reflection.DesktopPluginManager.<>c.<ResetServices>b__30_5()
   at OpenTabletDriver.Desktop.Reflection.ServiceManager.GetService(Type serviceType)
   at OpenTabletDriver.Desktop.Reflection.ServiceManager.GetService[T]()
   at OpenTabletDriver.Desktop.Profiles.AbsoluteModeSettings.GetDefaults(DigitizerSpecifications digitizer)
   at OpenTabletDriver.Desktop.Profiles.Profile.GetDefaults(TabletReference tablet)
   at OpenTabletDriver.Desktop.Profiles.ProfileCollection.<>c.<.ctor>b__2_0(TabletReference s)
   at System.Linq.Utilities.<>c__DisplayClass2_0`3.<CombineSelectors>b__0(TSource x)
   at System.Linq.Enumerable.SelectIListIterator`2.MoveNext()
   at System.Collections.Generic.List`1..ctor(IEnumerable`1 collection)
   at System.Collections.ObjectModel.ObservableCollection`1.CreateCopy(IEnumerable`1 collection, String paramName)
   at System.Collections.ObjectModel.ObservableCollection`1..ctor(IEnumerable`1 collection)
   at OpenTabletDriver.Desktop.Profiles.ProfileCollection..ctor(IEnumerable`1 profiles)
   at OpenTabletDriver.Desktop.Profiles.ProfileCollection..ctor(IEnumerable`1 tablets)
   at OpenTabletDriver.Desktop.Settings.GetDefaultProfiles()
   at OpenTabletDriver.Desktop.Settings.GetDefaults()
   at OpenTabletDriver.Daemon.DriverDaemon.ResetSettings()
   at OpenTabletDriver.Daemon.DriverDaemon.LoadUserSettings()
   at System.Threading.Tasks.Task.<>c.<ThrowAsync>b__128_1(Object state)
   at System.Threading.QueueUserWorkItemCallbackDefaultContext.Execute()
   at System.Threading.ThreadPoolWorkQueue.Dispatch()
   at System.Threading.PortableThreadPool.WorkerThread.WorkerThreadStart()
   at System.Threading.Thread.StartCallback()
sys_XOpenDisplay

Shizara, 2023-05-08:
> Logs from the daemon please.

5/8/2023 6:43:29 PM
System.DllNotFoundException
Unable to load shared library 'libX11' or one of its dependencies. In order to help diagnose loading problems, consider setting the LD_DEBUG environment variable: liblibX11: cannot open shared object file: No such file or directory
OpenTabletDriver.Native
   at OpenTabletDriver.Native.Linux.Xorg.XLib.sys_XOpenDisplay(Char* display)
   at OpenTabletDriver.Native.Linux.Xorg.XLib.XOpenDisplay(Char* display)
   at OpenTabletDriver.Desktop.Interop.Display.XScreen..ctor()
   at OpenTabletDriver.Desktop.Interop.DesktopInterop.ConstructLinuxDisplay()
   at OpenTabletDriver.Desktop.Interop.DesktopInterop.get_VirtualScreen()
   at OpenTabletDriver.Desktop.Reflection.DesktopPluginManager.<>c.<ResetServices>b__30_5()
   at OpenTabletDriver.Desktop.Reflection.ServiceManager.GetService(Type serviceType)
   at OpenTabletDriver.Desktop.Reflection.ServiceManager.GetService[T]()
   at OpenTabletDriver.Desktop.Profiles.AbsoluteModeSettings.GetDefaults(DigitizerSpecifications digitizer)
   at OpenTabletDriver.Desktop.Profiles.Profile.GetDefaults(TabletReference tablet)
   at OpenTabletDriver.Desktop.Profiles.ProfileCollection.<>c.<.ctor>b__2_0(TabletReference s)
   at System.Linq.Utilities.<>c__DisplayClass2_0`3.<CombineSelectors>b__0(TSource x)
   at System.Linq.Enumerable.SelectIListIterator`2.MoveNext()
   at System.Collections.Generic.List`1..ctor(IEnumerable`1 collection)
   at System.Collections.ObjectModel.ObservableCollection`1.CreateCopy(IEnumerable`1 collection, String paramName)
   at System.Collections.ObjectModel.ObservableCollection`1..ctor(IEnumerable`1 collection)
   at OpenTabletDriver.Desktop.Profiles.ProfileCollection..ctor(IEnumerable`1 profiles)
   at OpenTabletDriver.Desktop.Profiles.ProfileCollection..ctor(IEnumerable`1 tablets)
   at OpenTabletDriver.Desktop.Settings.GetDefaultProfiles()
   at OpenTabletDriver.Desktop.Settings.GetDefaults()
   at OpenTabletDriver.Daemon.DriverDaemon.ResetSettings()
   at OpenTabletDriver.Daemon.DriverDaemon.LoadUserSettings()
   at System.Threading.Tasks.Task.<>c.<ThrowAsync>b__128_1(Object state)
   at System.Threading.QueueUserWorkItemCallbackDefaultContext.Execute()
   at System.Threading.ThreadPoolWorkQueue.Dispatch()
   at System.Threading.PortableThreadPool.WorkerThread.WorkerThreadStart()
   at System.Threading.Thread.StartCallback()
sys_XOpenDisplay

jamesbt365, 2023-05-08:
Install libx11-dev and libxrandr-dev, may have a different name depending on the distro, you are missing the dependencies otd needs.

closed 2023-05-08 by jamesbt365