OpenTabletDriver/OpenTabletDriver #2743: OpenTabletDriver doesn't detects Genius i405x even though it says on the website it's supported

p0c0k, 2023-05-20:
### Acknowledgements

- [X] I have installed OpenTabletDriver by following its [official installation instructions](https://opentabletdriver.net/).
- [X] I have checked the [OpenTabletDriver Wiki](https://opentabletdriver.net/Wiki) and respective FAQ pages, and my issue was not covered or actually fixed.
- [X] I have searched the existing issues and this new issue is not a duplicate of any.
- [X] I have written a concise and meaningful title.

### Steps to Reproduce

1. Open OpenTabletDriver
2. No Tablets are detected.

### Expected Behavior

According to the website it's supported

### Observed Behavior

No Tablets are detected.

### Tablet Device

Genius i405x

### Diagnostic Information

{
  "App Version": "OpenTabletDriver v0.6.0.3",
  "Build Date": "12/30/2022",
  "Operating System": {
    "Platform": 4,
    "ServicePack": "",
    "Version": "6.2.14.300",
    "VersionString": "Unix 6.2.14.300"
  },
  "Environment Variables": {
    "USER": "p0c0k",
    "TEMP": null,
    "TMP": null,
    "TMPDIR": null,
    "DISPLAY": ":0",
    "WAYLAND_DISPLAY": "wayland-0",
    "PWD": "/home/p0c0k",
    "PATH": "/home/p0c0k/.local/bin:/home/p0c0k/bin:/usr/local/bin:/usr/local/sbin:/usr/bin:/usr/sbin:/home/p0c0k/.dotnet/tools"
  },
  "HID Devices": [
    {
      "DevicePath": "/sys/devices/pci0000:00/0000:00:08.1/0000:09:00.3/usb3/3-2/3-2.3/3-2.3:1.0/0003:145F:01BC.0002/hidraw/hidraw1",
      "Manufacturer": "Unknown Manufacturer",
      "ProductName": "USB Gaming Mouse",
      "SerialNumber": "",
      "FriendlyName": "USB Gaming Mouse",
      "VendorID": 5215,
      "ProductID": 444,
      "InputReportLength": -1,
      "OutputReportLength": -1,
      "FeatureReportLength": -1,
      "CanOpen": true
    },
    {
      "DevicePath": "/sys/devices/pci0000:00/0000:00:01.2/0000:02:00.0/usb1/1-8/1-8:1.0/0003:26CE:01A2.0008/hidraw/hidraw7",
      "Manufacturer": "ASRock",
      "ProductName": "LED Controller",
      "SerialNumber": "A02019100900",
      "FriendlyName": "LED Controller",
      "VendorID": 9934,
      "ProductID": 418,
      "InputReportLength": 65,
      "OutputReportLength": 65,
      "FeatureReportLength": 0,
[... 1467 lines cut ...]
      "Time": "2023-05-20T16:19:27.5184782+02:00",
      "Group": "Settings",
      "Message": "Driver is enabled.",
      "StackTrace": null,
      "Level": 1,
      "Notification": false
    },
    {
      "Time": "2023-05-20T16:19:27.5184894+02:00",
      "Group": "IPC",
      "Message": "Connected to a client.",
      "StackTrace": null,
      "Level": 0,
      "Notification": false
    },
    {
      "Time": "2023-05-20T16:19:27.5185007+02:00",
      "Group": "Settings",
      "Message": "Presets have been refreshed. Loaded 0 presets.",
      "StackTrace": null,
      "Level": 1,
      "Notification": false
    }
  ]
}

jamesbt365, 2023-05-20:
It is supported upstream but not currently in the latest version.

jamesbt365, 2023-08-23:
I assume this is the same variant that was added in #2238 (now in v0.6.1+) but can't 100% tell with the missing udev rules in the diagnostic above hiding the identifiers, please reopen if still not resolved on latest release.

closed 2023-08-23 by jamesbt365